arendst/Tasmota #3943: Dual R2 (esp8285) Failed Flash

howroyd, 2018-09-30:
I am trying to flash my Sonoff Dual R2 (esp8285) using the Arduino IDE.  I've flashed a few esp8266 devices already without issue but am struggling to find the correct settings.
Upload gets to 100%, appears to have worked but the device doesn't appear to boot. No flashy LED.

Settings I am currently using are in the image below.

Arduino IDE 1.8.2
Tasmota 6.2.1

[image]

Jason2866, 2018-09-30:
Dont use modul ESP8285! select ESP8266 and use exact same settings as you haved used for a ESP8266

Jason2866, 2018-09-30:
Flash mode DOUT is mandatory

howroyd, 2018-09-30:
Yep got DOUT set in my esp8266 settings.  Will try that now, thank you.  I'll also suggest an edit to the Wiki, it's not stated that you flash an esp8285 as an esp8266 and not obvious either.

howroyd, 2018-09-30:
Doesn't mention about selecting a module different to the chip you're using.  Maybe it's just me but I wouldn't flash an Arduino Nano with Arduino Mega selected in the tools.  DOUT isn't an option in the tools dropdown (image above) so I assumed [fail!] that this was default setting in the background.  I have no knowledge of compilers or uploaders I am afraid so wouldn't know how to check this.
Just trying to make things more obvious for other muppets like me!
Keep up the great work BTW, absolutely love the Tasmota project, very handy indeed.

Jason2866, 2018-09-30:
Please close this issue.
Thx

howroyd, 2018-09-30:
I will once I get it working, no luck yet with the suggested 8266 settings.  Trying to figure out whether its me or the settings at fault

howroyd, 2018-09-30:
It's having none of it:
`A fatal error occurred: Failed to write compressed data to flash after seq 3 (result was C100)`

Using the following command
`esptool.py --port COM3 --baud 115200 write_flash -fs 1MB -fm dout 0x0 sonoff-minimal.bin`

Seems to fail at a different % each time

Jason2866, 2018-09-30:
Seems like a hardware problem. We can talk at Discord Chat
https://discord.gg/Ks2Kzd4

closed 2018-09-30 by ascillato2

howroyd, 2018-10-01:
To close this off with a definitive answer for anyone else with a similar issue.  I bought a new FTDI dongle ("Moyina", on Amazon) and it's flashing first time with no issues.
Looks like the buspirate is temperamental when it comes to using it as an ISP.  Sometimes works, sometimes doesn't.
Hope this helps others!
Many thanks to everyone who provided me support with this, much appreciated.