arendst/Tasmota #3945: Max length hostname MQTT server to limited

jghaanstra, 2018-09-30:
**Describe the bug**
I want to use an MQTT server with a hostname of 34 characters but when saving it in the configuration it gets cut of to 33 characters resulting in the device not being able to connect to my MQTT server.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : Magic Home RGBW controller
- [ ] _Development/Compiler/Upload tools used_ : Atom with Platform.io
- [ ] _Provide the output of command_``status 0`` :
```
19:14:19 RSL: stat/sonoff/STATUS = {"Status":{"Module":34,"FriendlyName":["MagicHome"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:14:19 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:05","StartupUTC":"2018-09-30T18:11:14","Sleep":0,"BootCount":8,"SaveCount":20,"SaveAddress":"F8000"}}
19:14:19 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-30T19:27:47","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:14:19 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["***","***"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
19:14:19 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":467,"Free":536,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000413","0FCAE794","24000000","23B617CE","00000000"]}}
19:14:19 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1446","IPAddress":"192.168.0.144","Gateway":"192.168.0.100","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.100","Mac":"60:01:94:AA:25:A6","Webserver":2,"WifiConfig":5}}
19:14:19 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"diskstation.***.inf","MqttPort":1888,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA25A6","MqttUser":"magichome","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:14:19 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 30 18:14:19 2018","Local":"Sun Sep 30 19:14:19 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:33","Sunset":"18:13"}}
19:14:19 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-30T19:14:19"}}
19:14:19 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-30T19:14:19","Uptime":"0T00:03:05","Vcc":3.142,"POWER":"ON","Dimmer":100,"Color":"FFFFFF","HSBColor":"0,0,100","Channel":[100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"***","RSSI":54,"APMac":"54:A0:50:D8:52:A8"}}}
```

**To Reproduce**
Try to enter and save a hostname in MQTT settings with more than 33 characters. It saves but when you look at the log you can see it being cut off at 33 characters.

**Expected behavior**
Use unlimited lengthy hostnames.

**Additional context**
I had also set the hostname in user_config.h, I can even see it being the default value (as complete hostname) but when saving it still gets cut off.

ascillato2, 2018-09-30:
Hi,

Sorry, it is not supported

closed 2018-09-30 by ascillato2

Kaskadeur111, 2020-02-25:
Hi, i could also need this.
I am using 12 Power Plugs and connect them to iOBroker via MQTT.
The Hostname describes the Plugged in Devices (e.G. one Device Name is:  FritzBox, Raspi, Alexa, Switch, LGTV), and it can also be seen in iOBroker).
The MQTT Hostname ist also autamaticaly shown in iOBroker. But i cant use it, because its cut of at 33 letters. So i always need to change the Description in Tasmota AND in iOBroker if something changes. I Think 50 letters would be usefull (at least in my case).

arendst, 2020-02-25:
Upgrade to latest and try again.