arendst/Tasmota #3948: Rules using dimmer values

hank, 2018-09-30:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
I'd like to be able to write a rule that changes LEDs on a switch based upon the dimmer value.

**Describe the solution you'd like**
`rule3 on Dimmer1#state<30 do backlog power2 1; power3 0; power4 0 endon`
I'd also like to know how to enable trigger debug output in the weblog. It would be useful to know what triggers when I type `dimmer -` so I could hook that with a rule.

**Describe alternatives you've considered**
I've tried Dimmer1#state, Power6#Dimmer, and some other combinations but it never triggers.

**Additional context**
Finalizing implementation of the Martin Jerry wall dimmer. The status LEDs based on the dimming level is the last part before I can commit all of it and install it in my wall :)

hank, 2018-09-30:
I was also trying to get the 3 status LEDs to light up based on event values, but it was a) too slow to properly work when someone pushes a button fast, and b) wouldn't actually trigger the rule to change the LEDs.

```
rule1 on event#setdimmer do backlog dimmer %var1%; mem1 %var1%; event toggleleds=%var1% endon

rule2 on event#toggleleds>10.00 power2 1 endon on event#toggleleds>50.00 power3 1 endon on event#toggleleds>75.00 power4 1 endon on event#toggleleds<76.00 power4 0 endon on event#toggleleds<51.00 power3 0 endon on event#toggleleds<11.00 power2 0 endon

rule3 on switch2#state do backlog var1 %mem1%; sub1 10; event setdimmer endon on switch3#state do backlog var1 %mem1%; add1 10; event setdimmer endon
```
Any advice would be appreciated.

ascillato, 2018-10-01:
Hi,

There is no trigger for dimmer in rules. I also think that it is needed.

The actual available triggers for rules are at: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#trigger

I will try to do a PR, for Theo to review it, with a new trigger like:

`rule on Dimmer1#state<30 do ... endon`

Thanks for sharing your ideas.

hank, 2018-10-01:
That would be amazing - thanks!

ascillato, 2018-10-01:
Proposed the PR #3953 with Dimmer trigger for rules

closed 2018-10-01 by ascillato2