arendst/Tasmota #3951: Toggled rele on every reboot

vladspbru, 2018-10-01:
**Describe the bug**
I'm using Sonoff Base, Sonoff RF to switch contactors. Every time , when sonoff reboot, short-term impulse appears on output. 
In the next log, every 3 min, i can hear how the relay clicks:

00:03:16 APP: Restarting

16:12:40.430 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,6)
16:12:40.430 -> 
16:12:40.430 -> load 0x4010f000, len 1384, room 16 
16:12:40.430 -> tail 8
16:12:40.430 -> chksum 0x2d
16:12:40.430 -> csum 0x2d
16:12:40.463 -> v3fff3388
16:12:40.463 -> ~ld

00:00:00 Project sonoff Sonoff (Topic sonoff_68AC8D, Fallback DVES_68AC8D, GroupTopic sonoffs) Version 6.2.1.8-2_3_0
00:00:00 WIF: Connecting to AP1 karelia in mode 11N as sonoff_68AC8D-3213...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 garden in mode 11N as sonoff_68AC8D-3213...
00:00:14 WIF: Connect failed as AP cannot be reached
16:12:54.686 -> 00:00:14 WIF: WifiManager active for 3 minutes
00:00:14 HTP: Web server active on sonoff_68AC8D-3213 with IP address 192.168.4.1
00:02:00 RSL: tele/sonoff_68AC8D/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:01"}
00:03:16 APP: Restarting

16:15:56.815 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,6)
16:15:56.815 -> 
16:15:56.815 -> load 0x4010f000, len 1384, room 16 
16:15:56.815 -> tail 8
16:15:56.815 -> chksum 0x2d
16:15:56.815 -> csum 0x2d
16:15:56.848 -> v3fff3388
16:15:56.848 -> ~ld

00:00:00 Project sonoff Sonoff (Topic sonoff_68AC8D, Fallback DVES_68AC8D, GroupTopic sonoffs) Version 6.2.1.8-2_3_0
00:00:00 WIF: Connecting to AP2 garden in mode 11N as sonoff_68AC8D-3213...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 karelia in mode 11N as sonoff_68AC8D-3213...
00:00:14 WIF: Connect failed as AP cannot be reached
16:16:11.102 -> 00:00:14 WIF: WifiManager active for 3 minutes
00:00:14 HTP: Web server active on sonoff_68AC8D-3213 with IP address 192.168.4.1
00:02:00 RSL: tele/sonoff_68AC8D/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:01"}
00:03:16 APP: Restarting

STATUS 0 OUTPUT HERE
```

00:00:02 CMD: status 0
16:25:48.633 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS = {"Status":{"Module":2,"FriendlyName":["Sonoff"],"Topic":"sonoff_68AC8D","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:25:48.633 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:02","StartupUTC":"","Sleep":0,"BootCount":155,"SaveCount":323,"SaveAddress":"F5000"}}
16:25:48.672 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS2 = {"StatusFWR":{"Version":"6.2.1.8","BuildDateTime":"2018-09-27T14:20:41","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:25:48.672 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["karelia","garden"],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
16:25:48.699 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS4 = {"StatusMEM":{"ProgramSize":477,"Free":524,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","240003A1","23B617CE","000003C0"]}}
16:25:48.732 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS5 = {"StatusNET":{"Hostname":"sonoff_68AC8D-3213","IPAddress":"0.0.0.0","Gateway":"192.168.43.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.43.1","Mac":"5C:CF:7F:68:AC:8D","Webserver":2,"WifiConfig":3}}
16:25:48.732 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS6 = {"StatusMQT":{"MqttHost":"domus","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_68AC8D","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:25:48.765 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:02 1970","Local":"Thu Jan 01 00:00:02 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
16:25:48.798 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:02"}}
16:25:48.798 -> 00:00:02 RSL: stat/sonoff_68AC8D/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:02","Uptime":"0T00:00:02","Vcc":3.190,"POWER":"ON","Wifi":{"AP":1,"SSId":"karelia","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100}}}

**To Reproduce**
Flash tasmota on sonoff. Boot without network, hear how the relay clicks

**Expected behavior**
No rele click on boot.

ascillato, 2018-10-01:
> when sonoff reboot, short-term impulse appears on output.

Yes, you are right. It is a known issue. The ESP8266 starts with most of its GPIOs as low and then, after booting, it power them up as saved in the memory.

Sorry, we were not able to fix it by now. It is a esp8266 boot-time issue.

closed 2018-10-01 by vladspbru

andrethomas, 2018-10-01:
@vladspbru The clicking of the relay is expected behavior for Tasmota on sonoff devices.

The 3 minute interval is because the device is rebooting because you have not yet configured the wifi credentials. As soon as you configure the wifi credentials the 3 minute reboot will stop.

You will then only have a relay debounce when you reboot the sonoff (whilst the relay is on), or if you do a firmware upgrade (whilst the relay is on)

vladspbru, 2018-10-01:
there may be an option, disabling a hard reboot, or soft reboot, or config timeout to reboot on lack of a network

ascillato, 2018-10-01:
yes, config your wifi credentials and after that go to console and type `wificonfig 5` in order to not reboot when losing wifi connection.

andrethomas, 2018-10-01:
Try command
WifiConfig 5
From the wiki here https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

vladspbru, 2018-10-01:
thanks for the help

andrethomas, 2018-10-01:
@vladspbru You're welcome. Hope it works for you now.