arendst/Tasmota #3956: Michael Haustein ESP Switch

h-tro, 2018-10-01:
Tasmota 6.2.1.10:
Add support for Michael Haustein ESP Switch

Question:
What is a Michael Haustein ESP Switch?
Is there a link to it?

andrethomas, 2018-10-01:
Appears to be just another 4 relay board controlled directly by GPIO from what I can tell. Will have to wait for @arendst to respond as I don't find any specific reference to it other than the template created in the development codebase.

```
 { "ESP Switch",      // Michael Haustein 4 channel wall switch (ESP07 = ESP8266)
                       // Use rules for further actions like - rule on power1#state do publish cmnd/other_device/power %value% endon
     GPIO_KEY2,        // GPIO00 Button 2
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_REL3_INV,    // GPIO02 Yellow Led 3 (0 = On, 1 = Off)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_KEY1,        // GPIO04 Button 1
     GPIO_REL2_INV,    // GPIO05 Red Led 2 (0 = On, 1 = Off)
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_REL4_INV,    // GPIO12 Blue Led 4 (0 = On, 1 = Off)
     GPIO_KEY4,        // GPIO13 Button 4
     GPIO_KEY3,        // GPIO14 Button 3
     GPIO_LED1,        // GPIO15 Optional sensor
     GPIO_REL1_INV,    // GPIO16 Green Led 1 (0 = On, 1 = Off)
},
```

ascillato, 2018-10-01:
Hi,

For general questions, please use [Tasmota Discord Chat](https://discord.gg/Ks2Kzd4).

Thanks.

arendst, 2018-10-01:
I asked Michael for some info. In short it's a nice box with four switches and leds configured as relays. The idea is to use the buttons to initiate something.

andrethomas2, 2018-10-01:
Closing this issue as it has been answered - if you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-01 by andrethomas2

drivelog, 2018-10-03:
Infos to the Wallswitch will follow. It is a 71x71x25 mm WallCase that I have used to build a "ESP Wallswitch" with 4 Buttons and 4 corresponding LED's in different Colors. You can use it to control Lights via MQTT for example.... Regards Michael Haustein

drivelog, 2018-10-03:
It is still in Prototype phase and I will add a DS18B20 and an Epaper for descriting the Buttons...
The LED's I will change to WS2812 to save IO's and for more possibilities with the LED Colors...
MH

andrethomas, 2018-10-03:
@drivelog Sounds good - any photos yet? Also, will you break out something like I2C for expansion purposes?

drivelog, 2018-10-03:
Yes, all GPIO will go to a pad or header for other universal extension access.

drivelog, 2018-10-03:
I plan to set a DS2483 on the PCB - it is a 1-Wire Busmaster....usefull for long and bad 1 Wire Lines, like 3x1.5 NYM

drivelog, 2018-10-03:
Some Photos
[image]
[image]
[image]

drivelog, 2018-10-03:
[image]
[image]
[image]

andrethomas, 2018-10-04:
Hell yeah, I want some of those ;)

PuceBaboon, 2018-10-08:
Interesting wall box; where do you get them?

drivelog, 2018-10-08:
It is selfmade, I will build some in November 2018 as a small Serie.