arendst/Tasmota #3957: Sonoff Touch/T1 button light and action

aimartin, 2018-10-01:
Hi all,

I've looked for this in the documentation, but I couldn't find a way to do it.

I'd like to have the Sonoff Touch and T1 light as in the Sonoff switches with switchmode 2, where the light of the button is ON when the switch is OFF, so I can see the button in the dark.

Apart from it, I think there is no way to avoid the switch to be actioned when the button is pressed, I would like a way to configure the sonoff touch to send a MQTT message when the button is pressed instead of toggling the switch, and toggling the switch using MQTT directly.

Thanks

ascillato, 2018-10-01:
Hi,

> I'd like to have the Sonoff Touch and T1 light as in the Sonoff switches with switchmode 2, where the light of the button is ON when the switch is OFF, so I can see the button in the dark.

You will need to use generic module and set GPIOs as the T1 template.
Then you will need to add rules for managing your leds (as if they were relays)

> I think there is no way to avoid the switch to be actioned when the button is pressed

There is a way. That can be done using rules. See example https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

With all this, you should achieve the behaviour you want.

If you need help or further assistance with the rules, please, just ask. No problem at all

ascillato, 2018-10-01:
As the template in the code is:

```
  { "Sonoff T1 1CH",   // Sonoff T1 1CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Optional Sensor (J3 Pin 5)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0, 0,
     0, 0, 0,          // Flash connection
     0, 0,
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },
```

May be it is easier if you set the module GENERIC and set the GPIOs as:
```
GPIO0 SWITCH1
GPIO12 RELAY1
GPIO13 LED1
```

Please, test that if makes the behaviour you want regarding to the LED

For the switch and the MQTT:
go to the console and type
```
rule1 1
rule1 on switch1#state do publish cmnd/SONOFF_TOPIC/power %value% endon
```

ascillato, 2018-10-01:
If you need help or further assistance please, just ask. No problem at all.

andrethomas2, 2018-10-01:
Closing this issue as it has been answered - if you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-01 by andrethomas2

aimartin, 2018-10-02:
Thanks a lot, I'll test it

Cheers

Hurricane208, 2019-01-09:
From the Console I have got this much working:

switchmode1 3      //Disconnects  button1 from relay 1
rule1 1                   // enables rules
rule on Button1#State=2 do backlog power1 2; power2 2 endon  // Press button 1 to toggle switches 1 and 2

I plan to run switchmode2 3 so that I can use Button2 to toggle a Sonoff Basic

Now for the LEDs. 

> May be it is easier if you set the module GENERIC and set the GPIOs as:
> 
> GPIO0 SWITCH1
> GPIO12 RELAY1
> GPIO13 LED1
> Please, test that if makes the behaviour you want regarding to the LED
> 

@ascillato Can you help me further or direct me to where I can read more about this. Unfortunately I don't know what setting the module "generic" means.

Thank you for your help so far.

ascillato, 2019-01-10:
>  I don't know what setting the module "generic" means.

In the Tasmota Web Menu, go to configuration, then configuration module, then choose module Generic on the dropdown box.

madradrakie, 2019-12-13:
did you worked it out? is it working in your situation? can you show me your settings?