arendst/Tasmota #3961: Display SWITCH status in GUI

poldim, 2018-10-02:
**Describe the solution you'd like**
Currently:
- When you select a relay, the relay's value (status) and control are added to the GUI (IP of device, homepage).
- When you select a sensor, the sensor's value is added to the GUI (IP of device, homepage).
- When you select a switch, the switch's value is **_NOT_** added to the GUI (IP of device, homepage).

**GUI Mockup:**
[image]

**Describe alternatives you've considered**
Currently, Tasmota can report the status of the switch over MQTT using rules.  This is great for it to be displayed elsewhere, but when troubleshooting and testing, it would be great if it was on the Tasmota GUI.

ascillato, 2018-10-02:
Hi,

Sorry, that is not supported and likely not to be added in order to keep flash and RAM memories as low as possible. Tasmota is intended to be used with a Home Automation Software. There, you can have displayed all the information that you require.

andrethomas, 2018-10-02:
> when troubleshooting and testing, it would be great if it was on the Tasmota GUI

This is not the core intention of the web ui as indicated by @ascillato 

If you want more debugging information use the serial or web console - If you want more debugging information there is always seriallog 3 and weblog 3 to increase the verboseness of the logging produced in the two console options.

Also, in addition to this there is also the status 0 command which will give you just about everything that is set - For more on this please see the various options for the Status command in the wiki here https://github.com/arendst/Sonoff-Tasmota/wiki/Commands - It could be that you only want selective information from status so a specific status X parameter may be more specific to your requirements at any given time.

andrethomas2, 2018-10-02:
Closing this issue as it has been answered.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-02 by andrethomas2