arendst/Tasmota #3963: Crash on Domotics scene/group message on MQTT

mycroes, 2018-10-02:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Sonota + manual upgrade through web UI
- [x] Hardware used : Sonoff Basic
- [x] Provide the output of command ``status 0`` :
```
19:54:54 MQT: stat/sonoff4/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff4","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:54:54 MQT: stat/sonoff4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Exception","Uptime":"0T00:09:30","StartupUTC":"2018-10-02T18:45:24","Sleep":0,"BootCount":34,"SaveCount":158,"SaveAddress":"F7000"}}
19:54:54 MQT: stat/sonoff4/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:54:54 MQT: stat/sonoff4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Konijntjes","Konijntjes"],"TelePeriod":300,"SetOption":["54000009","55818000","00000000"]}}
19:54:54 MQT: stat/sonoff4/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
19:54:54 MQT: stat/sonoff4/STATUS5 = {"StatusNET":{"Hostname":"sonoff4-5723","IPAddress":"192.168.178.53","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"62.179.104.196","Mac":"5C:CF:7F:71:D6:5B","Webserver":2,"WifiConfig":2}}
19:54:54 MQT: stat/sonoff4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.241","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_71D65B","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:54:54 MQT: stat/sonoff4/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 02 18:54:54 2018","Local":"Tue Oct 02 19:54:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:51","Sunset":"18:27"}}
19:54:54 MQT: stat/sonoff4/STATUS10 = {"StatusSNS":{"Time":"2018-10-02T19:54:54"}}
19:54:54 MQT: stat/sonoff4/STATUS11 = {"StatusSTS":{"Time":"2018-10-02T19:54:54","Uptime":"0T00:09:30","Vcc":3.154,"POWER1":"ON","Wifi":{"AP":2,"SSId":"Konijntjes","RSSI":90,"APMac":"54:67:51:DE:C7:AD"}}}
```
Output after crash:
```
00:00:00 Project sonoff Sonoff (Topic sonoff4, Fallback DVES_71D65B, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: Connecting to AP2 Konijntjes in mode 11N as sonoff4-5723...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff4-5723.local with IP address 192.168.178.53
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff4/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff4/POWER = 
00:00:05 MQT: tele/sonoff4/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.1","FallbackTopic":"DVES_71D65B","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff4/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff4-5723","IPAddress":"192.168.178.53"}
00:00:05 MQT: tele/sonoff4/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4022f47c epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
00:00:05 MQT: stat/sonoff4/RESULT = {"POWER1":"ON"}
00:00:05 MQT: stat/sonoff4/POWER1 = ON
00:00:06 MQT: domoticz/in = {"idx":2,"nvalue":1,"svalue":"","Battery":55,"RSSI":8}
19:45:38 MQT: tele/sonoff4/STATE = {"Time":"2018-10-02T19:45:38","Uptime":"0T00:00:14","Vcc":3.156,"POWER1":"ON","Wifi":{"AP":2,"SSId":"Konijntjes","RSSI":80,"APMac":"54:67:51:DE:C7:AD"}}
```

I'm using **sonoff.bin** version 6.2.1 (manual download from releases page). I have 3 devices that are working fine individually, but since I created a screne (type group) to group them in Domoticz one of the devices seems to have issues. I subscribed to `domoticz/out` with MQTT.fx to see what messages are being send and noticed Domoticz also sends a message for the scene/group itself. The body of the message is as follows:
```
{
   "LastUpdate" : "2018-10-02 20:39:45",
   "Name" : "Sfeerverlichting",
   "Status" : "Off",
   "Timers" : "true",
   "Type" : "Group",
   "idx" : "2"
}
```
Sending this message manually (using MQTT.fx) causes an identical restart of the Sonoff.

I'm not sure if it's normal (this is my first Domoticz install, I'm running a Docker container and had no issues at all setting it up), but **idx 2** is used both by the scene/group and the Sonoff. I added my Sonoffs first, so they have idx 1, 2, 3 and 4 (of which one is having issues since I started using it, but that's out of the scope of this issue). I added a scene first and then the group, so probably the scene had **idx 1** before I deleted it, and the group has **idx 2**.

Is it absolutely wrong that the scene/group has the same idx as the Sonoff (and do you perhaps have any clue how that could have happened), or is there something more I should have configured to prevent the exception?

arendst, 2018-10-02:
Nice problem. Will investigate tomorrow.

mycroes, 2018-10-03:
@arendst Is that by any chance something I could fix as newbie to ESP8266? I'm a C# developer (professionally) but I have a fondness for programming in general (and have done more than just C# in the past as well). I must admit though that I haven't even installed PlatformIO, never programmed my Arduino Uno, but this doesn't sound like a hard issue to handle...

arendst, 2018-10-03:
Me too. Didn't find the time yet.

mycroes, 2018-10-03:
Please correct me if I'm wrong, but I think the following could fix it:
Line 182 original:
```
if ((idx > 0) && (nvalue >= 0) && (nvalue <= 15)) {
```

proposed:
```
if ((idx > 0) && nvalue.success() && (nvalue >= 0) && (nvalue <= 15)) {
```

I have no clue how to build or anything, but this seems to make sense after checking the ArduinoJson source.

mycroes, 2018-10-03:
Err that won't work, `nvalue` is `int16_t`. However `JsonObject[<string>]` should return a JsonObjectSubscript which has `.success()` and templated conversion (is that what it's called in C++?).

arendst, 2018-10-04:
Thx for your suggestions.

Just released the fix. Pls test and if OK just close this issue.

ascillato2, 2018-10-04:
@mycroes 

Hi

Have you managed to test it?

mycroes, 2018-10-05:
No sorry, didn't get to it yesterday... I'll test it soon and I'll close the issue then (from the code change it seems fixed).

mycroes, 2018-10-05:
Can't confirm it works yet. Downloaded sonoff.bin from 4208bad from thehackbox.org/tasmota/, but now the sonoff is unreachable over network most of the time. Sometimes the web UI works for something like half a minute, but it's kinda unusable now.

I guess I have to solder header pins to update it using esptool, it didn't work the previous time when I attempted to wire it up using a breadboard and loose connections.

ascillato2, 2018-10-06:
Hi,

When you have time, please tell us if this have work for you. If not, remember that you can find us in the Tasmota Discord Chat.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-06 by ascillato2

mycroes, 2018-10-06:
So I soldered the header, flashed 6.2.1 back in, everything seemed to be fine (didn't test extensively, but web interface was working). Then flashed the above mentioned sonoff.bin, same issue. Web interface works sometimes, but most of the times a single request is all it'll handle. Went back to 6.2.1 again, everything fine so far.

It's a Sonoff Basic, I'm not sure if it ought to be able to just use the sonoff.bin or if that could be too much for it's limited resources already, but for now that's all I could try. I can try to do a custom build with some features disabled, but honestly I would be kinda disappointed if that's what it takes to run newer Tasmota versions on my Sonoffs.