arendst/Tasmota #3964: HomeKit

ageorgios, 2018-10-03:
Reopening issue with HomeKit since there is great implementations and open sourced code for it.
I already have HomeKit compatibe sonoffs, I would love to see it integrated with Tasmota!

https://github.com/maximkulkin/esp-homekit
https://github.com/maximkulkin/esp-homekit-demo

https://github.com/arendst/Sonoff-Tasmota/issues/3876

andrethomas, 2018-10-03:
The firmware you point to is allowing HomeKit to talk to a sonoff device (as in the hardware)

What you are asking is that Tasmota firmware should also emulate being a HomeKit device (i.e. emulation)

Tasmota is primarily MQTT driven so this is unlikely to be implemented in such a way that it emulates a HomeKit device.

There also appears to be an alternative route - for example, node-red seems to have a plugin that allows integration with HomeKit so that way you could run Tasmota on sonoff devices and still control them using HomeKit via a HA solution.

ageorgios, 2018-10-03:
Of course you can already integrate Tasmota to Homekit with https://github.com/nfarina/homebridge but the case with esp-homekit is native homekit on esp.

andrethomas, 2018-10-03:
Right, so confirmed there is a work around for this in at least one or two methods so that immediately decreases the priority of this request.