arendst/Tasmota #3965: Crashes nightly on T1 UK 3ch

nhorvath, 2018-10-03:
Almost every night my T1 UK 3ch will lock up and turn all the lights on. It will remain unresponsive until power is removed and restored. This seems to be happening more frequently now. Previously it was only once a month or so. It's possible there are external factors which have increased the frequency but I would like some help identifying what it might be.

This issue has persisted across version upgrades including effecting 5.1 (I'm running 6.2.1 now). It also persisted across me doing a config reset by incrementing the number.

I setup external syslog to try and capture what is happening. With level set to "More Debug" this is the last contents of the logs before it locks up. The last line is always MQTT so far.
```
Oct  3 04:31:32 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  3 04:31:32 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  3 04:31:41 sonoff-front-lights-4529 ESP-MQT: tele/sonoff-front-lights/STATE = {"Time":"2018-10-03T04:31:41","Uptime":"0T05:40:15","Vcc":3.499,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"redacted","RSSI":96,"APMac":"redacted"}}
Oct  3 04:31:41 sonoff-front-lights-4529 ESP-MQT: tele/sonoff-front-lights/SENSOR = {"Time":"2018-10-03T04:31:41","Switch4":"OFF"}
Oct  3 04:31:52 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  3 04:31:52 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  3 04:32:12 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  3 04:32:12 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  3 04:32:32 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  3 04:32:32 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  3 04:32:52 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  3 04:32:52 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  3 04:33:00 sonoff-front-lights-4529 ESP-SRC: MQTT
```
It is set up with an AMS312 PIR sensor on GPIO2 in Switch4n mode with an external pull down resistor.
SwitchTopic is 1
SwitchMode4 is 1
The rest are defaults.

I run an MQTT server in Hass.io which controls the actions of the PIR sensor.

My user_config_override.h follows with passwords and ssids removed:
```
#ifdef WIFI_CONFIG_TOOL
#undef WIFI_CONFIG_TOOL
#endif
#define WIFI_CONFIG_TOOL       WIFI_WAIT      // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT, WIFI_SERIAL)

#ifdef WEB_PASSWORD
#undef WEB_PASSWORD
#endif
#define WEB_PASSWORD           "..."          // [WebPassword] Web server Admin mode Password for WEB_USERNAME (empty string = Disable)

#ifdef SYS_LOG_HOST
#undef SYS_LOG_HOST
#endif
#ifdef SYS_LOG_PORT
#undef SYS_LOG_PORT
#endif
#ifdef SYS_LOG_LEVEL
#undef SYS_LOG_LEVEL
#endif
#define SYS_LOG_HOST           "..."  // [LogHost] (Linux) syslog host
#define SYS_LOG_PORT           514                  // [LogPort] default syslog UDP port
#define SYS_LOG_LEVEL          LOG_LEVEL_INFO       // [SysLog] (LOG_LEVEL_NONE, LOG_LEVEL_ERROR, LOG_LEVEL_INFO, LOG_LEVEL_DEBUG, LOG_LEVEL_DEBUG_MORE)

#ifdef STA_SSID1
#undef STA_SSID1
#endif
#define STA_SSID1              "..."      // [Ssid1] Wifi SSID

#ifdef STA_PASS1
#undef STA_PASS1
#endif
#define STA_PASS1              "..."  // [Password1] Wifi password

#ifdef STA_SSID2
#undef STA_SSID2
#endif
#define STA_SSID2              "..."      // [Ssid2] Optional alternate AP Wifi SSID

#ifdef STA_PASS2
#undef STA_PASS2
#endif
#define STA_PASS2              "..."  // [Password2] Optional alternate AP Wifi password

#ifdef MQTT_HOST
#undef MQTT_HOST
#endif
#define MQTT_HOST            "hassio.horvath"  // [MqttHost]

#ifdef MQTT_PORT
#undef MQTT_PORT
#endif
#define MQTT_PORT            1883                // [MqttPort] MQTT TLS port

#ifdef MQTT_USER
#undef MQTT_USER
#endif
#define MQTT_USER            "sonoff"      // [MqttUser] Mandatory user

#ifdef MQTT_PASS
#undef MQTT_PASS
#endif
#define MQTT_PASS            "..."  // [MqttPassword] Mandatory password

#ifdef MQTT_POWER_RETAIN
#undef MQTT_POWER_RETAIN
#endif
#define MQTT_POWER_RETAIN      1                 // [PowerRetain] Power status message may send retain flag (0 = off, 1 = on)

// -- Time - Up to three NTP servers in your region
#ifdef NTP_SERVER1
#undef NTP_SERVER1
#endif
#ifdef NTP_SERVER2
#undef NTP_SERVER2
#endif
#ifdef NTP_SERVER3
#undef NTP_SERVER3
#endif
#define NTP_SERVER1            "pool.ntp.org"       // [NtpServer1] Select first NTP server by name or IP address 
#define NTP_SERVER2            "us.pool.ntp.org"    // [NtpServer2] Select second NTP server by name or IP address 
#define NTP_SERVER3            "0.us.pool.ntp.org"  // [NtpServer3] Select third NTP server by name or IP address

// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#ifdef TIME_DST_HEMISPHERE
#undef TIME_DST_HEMISPHERE
#endif
#ifdef TIME_DST_WEEK
#undef TIME_DST_WEEK
#endif
#ifdef TIME_DST_DAY
#undef TIME_DST_DAY
#endif
#ifdef TIME_DST_MONTH
#undef TIME_DST_MONTH
#endif
#ifdef TIME_DST_HOUR
#undef TIME_DST_HOUR
#endif
#ifdef TIME_DST_OFFSET
#undef TIME_DST_OFFSET
#endif
#define TIME_DST_HEMISPHERE    North              // [TimeDst] Hemisphere (0 or North, 1 or South)
#define TIME_DST_WEEK          Second             // Week of month (0 or Last, 1 or First, 2 or Second, 3 or Third, 4 or Fourth)
#define TIME_DST_DAY           Sun                // Day of week (1 or Sun, 2 or Mon, 3 or Tue, 4 or Wed, 5 or Thu, 6 or Fri, 7 or Sat)
#define TIME_DST_MONTH         Mar                // Month (1 or Jan, 2 or Feb, 3 or Mar, 4 or Apr, 5 or May, 6 or Jun, 7 or Jul, 8 or Aug, 9 or Sep, 10 or Oct, 11 or Nov, 12 or Dec)
#define TIME_DST_HOUR          2                  // Hour (0 to 23)
#define TIME_DST_OFFSET        -240               // Offset from UTC in minutes (-780 to +780)

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#ifdef TIME_STD_HEMISPHERE
#undef TIME_STD_HEMISPHERE
#endif
#ifdef TIME_STD_WEEK
#undef TIME_STD_WEEK
#endif
#ifdef TIME_STD_DAY
#undef TIME_STD_DAY
#endif
#ifdef TIME_STD_MONTH
#undef TIME_STD_MONTH
#endif
#ifdef TIME_STD_HOUR
#undef TIME_STD_HOUR
#endif
#ifdef TIME_STD_OFFSET
#undef TIME_STD_OFFSET
#endif
#define TIME_STD_HEMISPHERE    North              // [TimeStd] Hemisphere (0 or North, 1 or South)
#define TIME_STD_WEEK          First              // Week of month (0 or Last, 1 or First, 2 or Second, 3 or Third, 4 or Fourth)
#define TIME_STD_DAY           Sun                // Day of week (1 or Sun, 2 or Mon, 3 or Tue, 4 or Wed, 5 or Thu, 6 or Fri, 7 or Sat)
#define TIME_STD_MONTH         Nov                // Month (1 or Jan, 2 or Feb, 3 or Mar, 4 or Apr, 5 or May, 6 or Jun, 7 or Jul, 8 or Aug, 9 or Sep, 10 or Oct, 11 or Nov, 12 or Dec)
#define TIME_STD_HOUR          3                  // Hour (0 to 23)
#define TIME_STD_OFFSET        -300               // Offset from UTC in minutes (-780 to +780)

// -- Location ------------------------------------
#ifdef LATITUDE
#undef LATITUDE
#endif
#ifdef LONGITUDE
#undef LONGITUDE
#endif
#define LATITUDE               ...        // [Latitude] Your location to be used with sunrise and sunset
#define LONGITUDE            ...        // [Longitude] Your location to be used with sunrise and sunset

// -- Application ---------------------------------
#ifdef APP_TIMEZONE
#undef APP_TIMEZONE
#endif
#define APP_TIMEZONE           99                // [Timezone] +1 hour (Amsterdam) (-13 .. 14 = hours from UTC, 99 = use TIME_DST/TIME_STD)

#ifdef USE_DOMOTICZ
#undef USE_DOMOTICZ
#endif

#ifdef USE_I2C
#undef USE_I2C
#endif

#ifdef USE_IR_REMOTE
#undef USE_IR_REMOTE
#endif

#ifdef USE_WS2812
#undef USE_WS2812
#endif

#ifdef USE_EMULATION
#undef USE_EMULATION
#endif

#ifdef USE_DISCOVERY
#undef USE_DISCOVERY
#endif
```

ascillato, 2018-10-03:
Please, provide also the status 0

ascillato, 2018-10-03:
also you have a powerretain to 1. This causes issues also. see https://www.youtube.com/watch?v=31IyfM1gygo&t=2s

ascillato, 2018-10-03:
> Almost every night my T1 UK 3ch will lock up and turn all the lights on. It will remain unresponsive until power is removed and restored.

Seems to be a problem with your wifi core version but as there is no status 0 I can't know. If your esplib version is 2.4.0 or 2.4.1 that bug will occur.

nhorvath, 2018-10-03:
Here's status 0. Thanks for the help
```
10:30:10 MQT: stat/sonoff-front-lights/STATUS = {"Status":{"Module":30,"FriendlyName":["Entryway Lights","Front Door Lights","Porch Lights"],"Topic":"sonoff-front-lights","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T02:27:33","StartupUTC":"2018-10-03T12:02:37","Sleep":0,"BootCount":8,"SaveCount":552,"SaveAddress":"F4000"}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-30T06:17:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"...","LogPort":514,"SSId":["horvath","horvath5ghz"],"TelePeriod":300,"SetOption":["00008029","55818000","00000000"]}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS4 = {"StatusMEM":{"ProgramSize":450,"Free":552,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FC86580","00000001","23B617CE","00000000"]}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS5 = {"StatusNET":{"Hostname":"sonoff-front-lights-4529","IPAddress":"192.168.75.84","Gateway":"192.168.75.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.75.1","Mac":"...","Webserver":2,"WifiConfig":0}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.horvath","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_98B1B1","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 03 14:30:10 2018","Local":"Wed Oct 03 10:30:10 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 03:00:00 2018","Timezone":99,"Sunrise":"06:51","Sunset":"18:31"}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS10 = {"StatusSNS":{"Time":"2018-10-03T10:30:10","Switch4":"OFF"}}
10:30:10 MQT: stat/sonoff-front-lights/STATUS11 = {"StatusSTS":{"Time":"2018-10-03T10:30:10","Uptime":"0T02:27:33","Vcc":3.499,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"horvath","RSSI":100,"APMac":"..."}}}
```

nhorvath, 2018-10-03:
I watched the video about powerretain. It's not a ghost switch event. The device is locked up: no wifi access, buttons don't work, relays are all on but the leds behind the buttons are off.
Core is 2.4.2 so I don't think the 2.4.0-1 bug applies.

ascillato, 2018-10-03:
Using your status 0:
```
{"Status":{"Module":30,"FriendlyName":["Entryway Lights","Front Door Lights","Porch Lights"],"Topic":"sonoff-front-lights","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T02:27:33","StartupUTC":"2018-10-03T12:02:37","Sleep":0,"BootCount":8,"SaveCount":552,"SaveAddress":"F4000"},"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-30T06:17:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"...","LogPort":514,"SSId":["horvath","horvath5ghz"],"TelePeriod":300,"SetOption":["00008029","55818000","00000000"]},"StatusMEM":{"ProgramSize":450,"Free":552,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FC86580","00000001","23B617CE","00000000"]},"StatusNET":{"Hostname":"sonoff-front-lights-4529","IPAddress":"192.168.75.84","Gateway":"192.168.75.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.75.1","Mac":"...","Webserver":2,"WifiConfig":0},"StatusMQT":{"MqttHost":"hassio.horvath","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_98B1B1","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Wed Oct 03 14:30:10 2018","Local":"Wed Oct 03 10:30:10 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 03:00:00 2018","Timezone":99,"Sunrise":"06:51","Sunset":"18:31"},"StatusSNS":{"Time":"2018-10-03T10:30:10","Switch4":"OFF"},"StatusSTS":{"Time":"2018-10-03T10:30:10","Uptime":"0T02:27:33","Vcc":3.499,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"horvath","RSSI":100,"APMac":"..."}}}
```
with decode-status.py:
```
*** decode-status.py v20180730 by Theo Arends ***
Decoding information for device Entryway Lights from status report taken at 2018
-10-03T10:30:10

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (ON ) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) MQTT serial binary
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX enabled
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led
  50 (OFF) Timers enabled
  51 (OFF) Generic ESP8285 GPIO enabled
  52 (OFF) Add UTC time offset to JSON message

Features
  Language LCID = 2057
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  USE_ADC_VCC
  USE_ARILUX_RF
  USE_BH1750
  USE_BMP
  USE_CONFIG_OVERRIDE
  USE_DHT
  USE_DS18x20
  USE_ENERGY_SENSOR
  USE_HOME_ASSISTANT
  USE_HTU
  USE_IR_RECEIVE
  USE_LM75AD
  USE_MHZ19
  USE_NOVA_SDS
  USE_PMS5003
  USE_PZEM004T
  USE_RULES
  USE_SENSEAIR
  USE_SERIAL_BRIDGE
  USE_SGP30
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_WEBSERVER
  WEBSERVER_ADVERTISE
```

You can try to change the MQTT_PUBSUBCLIENT library to TASMOTA_MQTT in user_config.h file and recompile. Try to use lwIP 1.4

This alternative library is non blocking when you have a glitch or disconnection from your MQTT Broker.

nhorvath, 2018-10-03:
Come to think of it i used to compile with 1.4 and saw somewhere to switch to 2. It may have gotten worse since then. I'll try that. What are the main differences between mqtt_pubsubclient and tasmota_mqtt?

ascillato, 2018-10-03:
yes, depends on which libraries you use, sometimes 2 is better or worse

nhorvath, 2018-10-03:
Ok, all flashed up. Cross your fingers that it's not stuck tomorrow morning!

Jason2866, 2018-10-04:
If it still not working you can try to use the fixed IP instead of
`#define MQTT_HOST            "hassio.horvath"  // [MqttHost]`
for the mqtt broker. As Adrian wrote mDNS Discovery and MQTT Host Discovery sometimes makes trouble.

nhorvath, 2018-10-04:
Crashed again last night :(
@Jason2866 I'm not using discovery. It's just standard a DNS lookup.
It might be worth noting I use this exact firmware on all my other sonoffs, which are all basics though, and one nodemcu with no problems. Could it have something to do with the fact that T1 is an ESP8285? Do I need to change that when I compile firmware?

nhorvath, 2018-10-04:
Also, last night the last line of logs is not MQTT, just one of the periodic Checking connection... / Connected messages. And it should have sent another pair or two of those before the next tele message. Wifi signal was 98 in the last tele sent.

nhorvath, 2018-10-04:
And I just realized it crashed again after I reset it this morning, uptime of just over 5 mins this time.

```
Oct  4 06:29:55 sonoff-front-lights-4529 ESP-MQT: tele/sonoff-front-lights/STATE = {"Time":"2018-10-04T06:29:55","Uptime":"0T00:05:14","Vcc":3.499,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"horvath","RSSI":88,"APMac":"A0:04:60:C4:3D:4D"}}
Oct  4 06:29:55 sonoff-front-lights-4529 ESP-MQT: tele/sonoff-front-lights/SENSOR = {"Time":"2018-10-04T06:29:55","Switch4":"OFF"}
Oct  4 06:30:06 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:30:06 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:30:26 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:30:26 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:30:46 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:30:46 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:31:06 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:31:06 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:31:26 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:31:26 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:31:46 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:31:46 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:32:06 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:32:06 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:32:26 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:32:26 sonoff-front-lights-4529 ESP-WIF: Connected
Oct  4 06:32:46 sonoff-front-lights-4529 ESP-WIF: Checking connection...
Oct  4 06:32:46 sonoff-front-lights-4529 ESP-WIF: Connected
```

nhorvath, 2018-10-05:
I disabled my automations that automatically turn on the light based on a motion sensor outside (it's an exterior light) and it didn't crash last night.

ascillato, 2018-10-06:
@nhorvath 

Hi,

Can you give more details about that? Which Automation? from a Home Automation software or a rule? If it is an automation software, which one and which command is sending to your sonoff?

nhorvath, 2018-10-06:
No crash again last night. It's from Home Assistant. It's a kind of complicated rule that basically says If anyone got home (phone wifi tracking), or there's motion outside, and the sun is down and the lights are not already on, turn on the front door lights. Then 3 mins later turns them off. I did notice that one of the phone's wifi connections was flapping which was causing it to trigger often.
However, it couldn't trigger more than once every 3 mins. It is possible it could turn off and immediately retrigger though.

  command_topic: "cmnd/sonoff-front-lights/power2"
  payload_on: "ON"
  payload_off: "OFF"

ascillato2, 2018-10-06:
Have you tried to fire that command manually to see if that hungs your device?

nhorvath, 2018-10-06:
Oh yeah. It goes off all the time and is fine. It would just lock up randomly at night. I'm thinking maybe an issue with the number of times commands are sent?

ascillato, 2018-10-06:
The number of commands shouldn't be an issue. I have tested that and Tasmota works fine. No problem. If you send like 100 messages per second, a lot of messages will be ignored but will work.

ascillato, 2018-10-06:
Do you have anything else attached to your Sonoff? Or it is just standard ?

nhorvath, 2018-10-06:
This one has an AMS312 PIR sensor on GPIO2 in Switch4n mode with an external pull down resistor.
It's a 3 channel T1 (UK version)

ascillato, 2018-10-06:
The source voltage for the PIR is being taken from the sonoff or you have another power source for that?

I'm asking because the sonoff power source are very tight and taken power from them make some little power failures that gave some random issues.

nhorvath, 2018-10-06:
No separate power. VCC in the tele messages is always 3.500 or 3.499 though.

ascillato, 2018-10-06:
Ok, perfect. So, that is not an issue there.

ascillato, 2018-10-06:
So, to avoid any platformio issue, let's try this (I know it is a pain):

Erase all flash using esptool.py
Then upload with esptool.py the precompiled bin from http://thehackbox.org/tasmota/

nhorvath, 2018-10-06:
Oh i probably should have mentioned I flash with arduino the first time. After that I always use OTA.

ascillato, 2018-10-06:
ah, ok,

So as to discard any issue with incorrect information left in the flash, please, try this (I know it is a pain):

Erase all flash using esptool.py
Then upload with esptool.py the precompiled bin from http://thehackbox.org/tasmota/

ascillato2, 2018-10-07:
Hi

Have you managed to solve your issue?

nhorvath, 2018-10-07:
I haven't had a chance to reflash, but removing the phone that flaps
between home and away from the triggers have stopped the crashing. This
points to a problem with mqtt commands. I tried sending off/on commands
rapidly but could not get it to crash though. You can close this for now
unless that gives you any ideas.

ascillato, 2018-10-07:
I think that is better that you close the issue, so, if you start having the issue again you can reopen it. If we close it, you need to ask us for reopen.

FYI: We are now testing the new MQTT Library called ArduinoMQTT and it is very promising. Also the new core that will be released from Arduino (hope to be soon), should have a lot of issues solved including the sleep feature.

One thing I didn't ask you is that when you were compiling with Arduino, which lwIP did you choose? lwip 1.4, lwip 2 (low mem) or lwip 2 (high bandwidth) ?

the more stable is using core 2.3.0 or the 2.4.2 with the lwip 1.4 and using the TasmotaMQTT or ArduinoMQTT libraries.

nhorvath, 2018-10-07:
I was using iwip 2 high bandwith, but switched to iwip 1.4 and tasmota mqtt
at the suggestion of this thread. It didn't seem to make a difference.

closed 2018-10-07 by nhorvath