arendst/Tasmota #3966: Device reboot all time

123trs, 2018-10-03:
I use a Arilux LC10 with tasmota. It dont work out of the box, i have to rearrange the GPIOs like below:

Arilux LC11",     // Arilux AL-LC11 (ESP8266)
                       // https://www.banggood.com/nl/ARILUX-AL-LC11-Super-Mini-LED-WIFI-APP-Controller-RF-Remote-Control-For-RGBWW-LED-Strip-DC9-28V-p-1085112.html
                       //  (PwmFrequency 540Hz)
     GPIO_KEY1,        // GPIO00 Optional Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED2_INV,    // GPIO02 nur geraten
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_ARIRFRCV,        // GPIO004nur geraten
     GPIO_PWM4,        // GPIO005 nur geraten
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM2,        // GPIO12 RGBCW LED Warm
     GPIO_PWM3,        // GPIO13 RGBW LED Cold
     GPIO_PWM1,        // GPIO14 RGB LED Blue
     GPIO_PWM5,    // GPIO15 RF receiver input
     0, 0

Every works fine, the remote also. But the device reboots every minute:

00:00:00 Projekt sonoff Sonoff (Topic sonoffwohn, Fallback DVES_957AF7, GroupTopic sonoffs) Version 6.2.1.2-2_3_0
00:00:00 WIF: verbinden mit AP1 FRITZ!Box Fon WLAN 7390 TRS in Modus 11N wie sonoffwohn-6903...
00:00:04 WIF: verbunden
00:00:04 DNS: initialisiert
00:00:04 HTP: Web-Server aktiv bei sonoffwohn-6903.local mit IP-Adresse 192.168.1.92
00:00:05 MQT: Verbindungsversuch...
00:00:05 MQT: verbunden
00:00:05 MQT: /smarthome/wohn/sonoffwohn/tele/LWT = online (beibehalten)
00:00:05 MQT: /smarthome/wohn/sonoffwohn/cmnd/POWER = 
00:00:05 MQT: /smarthome/wohn/sonoffwohn/tele/INFO1 = {"Module":"Arilux LC11","Version":"6.2.1.2","FallbackTopic":"DVES_957AF7","GroupTopic":"sonoffs"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffwohn-6903","IPAddress":"192.168.1.92"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40210bd0 epc2:0x00000000 epc3:0x40229b8b excvaddr:0x00000000 depc:0x00000000"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/stat/RESULT = {"POWER":"OFF"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/stat/POWER = OFF
16:57:01 MQT: /smarthome/wohn/sonoffwohn/tele/STATE = {"Time":"2018-10-03T16:57:01","Uptime":"0T00:00:14","Vcc":3.208,"POWER":"OFF","Dimmer":89,"Color":"00E21F0000","HSBColor":"128,100,89","Channel":[0,89,12,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":80}}

I dont know why.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ :  Arilux LC11
- [ ] _Development/Compiler/Upload tools used_ : Atom/PlatformIO
- [ ] _Provide the output of command_``status 0`` :

16:59:43 CMD: status 0
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS = {"Status":{"Module":38,"FriendlyName":["Sonoff"],"Topic":"sonoffwohn","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:00:11","StartupUTC":"2018-10-03T14:59:32","Sleep":0,"BootCount":224,"SaveCount":856,"SaveAddress":"F9000"}}
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS2 = {"StatusFWR":{"Version":"6.2.1.2","BuildDateTime":"2018.10.03 12:13:48","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box Fon WLAN 7390 TRS",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","240003A0","23B617CE","000002C0"]}}
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoffwohn-6903","IPAddress":"192.168.1.92","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:95:7A:F7","Webserver":2,"WifiConfig":5}}
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_957AF7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:59:43 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 03 14:59:43 2018","Local":"Wed Oct 03 16:59:43 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:52","Sunset":"19:25"}}
16:59:44 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS10 = {"StatusSNS":{"Time":"2018-10-03T16:59:43"}}
16:59:44 MQT: /smarthome/wohn/sonoffwohn/stat/STATUS11 = {"StatusSTS":{"Time":"2018-10-03T16:59:44","Uptime":"0T00:00:12","Vcc":3.204,"POWER":"OFF","Dimmer":89,"Color":"00E21F0000","HSBColor":"128,100,89","Channel":[0,89,12,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":74}}}
16:59:48 MQT: /smarthome/wohn/sonoffwohn/tele/STATE = {"Time":"2018-10-03T16:59:48","Uptime":"0T00:00:16","Vcc":3.205,"POWER":"OFF","Dimmer":89,"Color":"00E21F0000","HSBColor":"128,100,89","Channel":[0,89,12,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":72}}
```
STATUS 0 OUTPUT HERE
```

**To Reproduce****IMPORTANT NOTICE**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

123trs, 2018-10-03:
If i use the generic modul, with only the PWM GPIOs, the device dont reboot. But then, the remote dont work. The problem is the remote.

andrethomas, 2018-10-03:
> GPIO_PWM5, // GPIO15 RF receiver input

Should that not be GPIO_USER since you want to use it as an input?

andrethomas, 2018-10-03:
Or GPIO_ARIRFRCV rather

andrethomas, 2018-10-03:
Right, the comment on the template you had " // GPIO15 RF receiver input" confused me ;)

123trs, 2018-10-03:
ok, commend is wrong. I dont update it. Only set the GPIO.

andrethomas, 2018-10-03:
Ok, I cannot reproduce it in the same way as you because I do not have the hardware but I'm guessing for some reason GPIO4 is causing the device to reset.

I see you say you're using platformio to compile - could you try a compile using Arduino just to rule it isn't one of those intermittent problems created by platformio ?

andrethomas, 2018-10-03:
Or upload your sonoff_template.h here and I'll compile it for you if you like.

123trs, 2018-10-03:
But without

GPIO_ARIRFRCV on GPIO04

Remote wont work.
I will send my sonoff_template;-)

123trs, 2018-10-03:
[sonoff_template.zip](https://github.com/arendst/Sonoff-Tasmota/files/2443418/sonoff_template.zip)

In german, please

ascillato, 2018-10-03:
may be the GPIO0 is pressed or may be is inverted in your hardware.

try to configure gpio0 as none and do not use key1 in any gpio. just to test.

andrethomas, 2018-10-03:
```
{ "Arilux LC11",     // Arilux AL-LC11 (ESP8266)
                       // https://www.banggood.com/nl/ARILUX-AL-LC11-Super-Mini-LED-WIFI-APP-Controller-RF-Remote-Control-For-RGBWW-LED-Strip-DC9-28V-p-1085112.html
                       //  (PwmFrequency 540Hz)
     GPIO_USER,        // GPIO00 Optional Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED2_INV,    // GPIO02 RF receiver control
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_ARIRFRCV,    // GPIO04 RGB LED Green
     GPIO_USER,        // GPIO05 RGB LED Red
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM2,        // GPIO12 RGBCW LED Warm
     GPIO_PWM3,        // GPIO13 RGBW LED Cold
     GPIO_PWM1,        // GPIO14 RGB LED Blue
     GPIO_PWM4,    // GPIO15 RF receiver input
     0, 0
  },
```

So here is this, let's test normal english first because the test is just to rule out potential platformio compile/linking issue.

[sonoff.zip](https://github.com/arendst/Sonoff-Tasmota/files/2443447/sonoff.zip)

andrethomas, 2018-10-03:
Do you have any other devices of the same type that does not produce this problem?

123trs, 2018-10-03:
No, two pices with the same behavior.

123trs, 2018-10-03:
the colours are wrong with the org. template and rc dont work.

andrethomas, 2018-10-03:
There are very few conditions that will cause a reset - the fact that this only happens when you use remote on GPIO4 means that GPIO4 is causing the reset - I cannot reproduce this on a bare esp8266 board so either you need to dig deeper to see where the pins are actually going on the hardware, or the receiver is pulling the VCC too low although the 3.208V seems fine.

So my conclusion is its either a hardware problem, or a wiring problem, or maybe at a long shot mDNS / DNS auto discovery is failing and causing a reset... but the latter does not explain why it will only happen if you use GPIO4... so back to wiring and/or hardware problem.

Jason2866, 2018-10-03:
Hello @123trs 
i think it is a hardware issue.  
Last try. Erase the device completly with esptool
Maybe fragements are left in flash which causes this error....

andrethomas, 2018-10-04:
Also, I don't think the receiver will work without AC because it most likely needs the 5V rail - did you test on AC after or just using the 3.3V from usb adapter

Jason2866, 2018-10-04:
@andrethomas  I agree.
Since it is safely DC Powered (9-28V) It should be flashed and tested with main DC Power connected

andrethomas, 2018-10-04:
Yeah, please try make sure the device is powered properly - it could be that the remote is pulling the VCC down too low causing a watchdog reset.

If the problem persists I will give you a new bin file and then we can see if we can trace the address where the exception comes from.

123trs, 2018-10-04:
Power unit is not the problen, i test it with an other one with the same result. I use the arilux on a 5m led stripe. The device reboot also with led turned off. Without any keypress.

123trs, 2018-10-04:
With the generic modul device runs, but i need the rc

Jason2866, 2018-10-04:
I think the chinese did a strange way of connecting the IR receiver...
Without the knowledge how the did this, help is just guessing around.

andrethomas, 2018-10-04:
Right, here's my last shot of hope.

Flash the attached binary and get the output of the console where it says the exception occurs...

With some luck, the linker map file will maybe point to where the exception is occurring.

[sonoff_bin.zip](https://github.com/arendst/Sonoff-Tasmota/files/2448383/sonoff_bin.zip)
[sonoff_map.zip](https://github.com/arendst/Sonoff-Tasmota/files/2448384/sonoff_map.zip)

123trs, 2018-10-05:
How does it work with the .map file?

123trs, 2018-10-05:
I will test it, and give you feedback, thanks.

123trs, 2018-10-05:
Sorry, cant flash: magic byte ist nicht 0xE9

andrethomas, 2018-10-05:
The sonoff.bin inside sonoff_bin.zip - I extracted and test flash - flashes fine for me.

123trs, 2018-10-05:
00:00:00 Project sonoff Arilux LC11 (Topic sonoffwohn, Fallback DVES_8D9C26, GroupTopic sonoffs) Version 6.2.1.11-2_3_0
00:00:00 WIF: Connecting to AP1 FRITZ!Box Fon WLAN 7390 TRS in mode 11N as sonoffwohn-7206...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoffwohn-7206.local with IP address 192.168.1.93
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: /smarthome/wohn/sonoffwohn/tele/LWT = Online (retained)
00:00:05 MQT: /smarthome/wohn/sonoffwohn/cmnd/POWER = 
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO1 = {"Module":"Arilux LC11","Version":"6.2.1.11","FallbackTopic":"DVES_8D9C26","GroupTopic":"sonoffs"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffwohn-7206","IPAddress":"192.168.1.93"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x402032cc epc2:0x00000000 epc3:0x40000f68 excvaddr:0x00000000 depc:0x00000000"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/stat/RESULT = {"POWER":"OFF"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/stat/POWER = OFF
21:03:47 MQT: /smarthome/wohn/sonoffwohn/tele/STATE = {"Time":"2018-10-05T21:03:47","Uptime":"0T00:00:14","Vcc":3.229,"POWER":"OFF","Dimmer":1,"Color":"02000000","HSBColor":"0,100,1","Channel":[1,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":84}}

andrethomas, 2018-10-05:
Are you using auto-discovery for MQTT?

andrethomas, 2018-10-05:
According to map file, exception is from
402032cc g     F .irom0.text	000000e0 _Z17AriluxRfInterruptv
So the exception is caused when the interrupt on that pin of the receiver is triggered and probably not serviced properly...

What are the lines output immediately after it resets... like ld something something

123trs, 2018-10-05:
I dont know, how you mean

andrethomas, 2018-10-05:
After the exception occurs, the device resets - it will output something similar to what I posted in my 2nd last post - what does yours look like?

123trs, 2018-10-05:
00:00:00 Project sonoff Arilux LC11 (Topic sonoffwohn, Fallback DVES_8D9C26, GroupTopic sonoffs) Version 6.2.1.11-2_3_0
00:00:00 WIF: Connecting to AP1 FRITZ!Box Fon WLAN 7390 TRS in mode 11N as sonoffwohn-7206...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoffwohn-7206.local with IP address 192.168.1.93
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: /smarthome/wohn/sonoffwohn/tele/LWT = Online (retained)
00:00:05 MQT: /smarthome/wohn/sonoffwohn/cmnd/POWER = 
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO1 = {"Module":"Arilux LC11","Version":"6.2.1.11","FallbackTopic":"DVES_8D9C26","GroupTopic":"sonoffs"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffwohn-7206","IPAddress":"192.168.1.93"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/tele/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x402032cc epc2:0x00000000 epc3:0x40000f68 excvaddr:0x00000000 depc:0x00000000"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/stat/RESULT = {"POWER":"OFF"}
00:00:06 MQT: /smarthome/wohn/sonoffwohn/stat/POWER = OFF
21:11:31 MQT: /smarthome/wohn/sonoffwohn/tele/STATE = {"Time":"2018-10-05T21:11:31","Uptime":"0T00:00:14","Vcc":3.228,"POWER":"OFF","Dimmer":1,"Color":"02000000","HSBColor":"0,100,1","Channel":[1,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":82}}
21:16:31 MQT: /smarthome/wohn/sonoffwohn/tele/STATE = {"Time":"2018-10-05T21:16:31","Uptime":"0T00:05:14","Vcc":3.227,"POWER":"OFF","Dimmer":1,"Color":"02000000","HSBColor":"0,100,1","Channel":[1,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":82}}
21:21:17 APP: Serial logging disabled
21:21:31 MQT: /smarthome/wohn/sonoffwohn/tele/STATE = {"Time":"2018-10-05T21:21:31","Uptime":"0T00:10:14","Vcc":3.230,"POWER":"OFF","Dimmer":1,"Color":"02000000","HSBColor":"0,100,1","Channel":[1,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":84}}
21:26:31 MQT: /smarthome/wohn/sonoffwohn/tele/STATE = {"Time":"2018-10-05T21:26:31","Uptime":"0T00:15:14","Vcc":3.226,"POWER":"OFF","Dimmer":1,"Color":"02000000","HSBColor":"0,100,1","Channel":[1,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7390 TRS","BSSId":"08:96:D7:49:BF:74","Channel":6,"RSSI":80}}

Thats all, or shall i set "SysLog 4"?

andrethomas, 2018-10-05:
No, what comes before that?

123trs, 2018-10-05:
Nothing, thats all into the console window

andrethomas, 2018-10-05:
So nothing before and nothing after? Do you have it powered from DC at the same time?

andrethomas, 2018-10-05:
I think you may be using the web console - i want the serial console output!

123trs, 2018-10-05:
No, nothing, and no power down. I have don nothing, no keypress, nothing. The device reboot, then run vor two or three minutes, and reboots again

123trs, 2018-10-05:
I dont know, how to see the console

andrethomas, 2018-10-05:
How are you uploading firmware? By serial? User a serial application to see what comes out the serial port.

123trs, 2018-10-05:
Over the web interface

123trs, 2018-10-05:
I must solder cables on the device, first

andrethomas, 2018-10-05:
So you never connected with cable before?

123trs, 2018-10-05:
Of course, to flash the bin. But i have done it with a pair of needles;-) I dont want to burn the device withe the iron... Works with a pair of Arilux lc01. But now, i need a device with 4(5) outputs vor rgbw stripes. It was also the first device with rc remote. The devices with ir remote runns beautiful

Jason2866, 2018-10-05:
So this is the point where a complete erase of the flash is neccersary!
Do that with esptool. Described in wiki. AFTER that flash the firmware @andrethomas 
has provided. 
Further investigation at this point is useless with a device with fragments of code in flash!