arendst/Tasmota #3969: Tasmota 6.2.1: When using cycling STA_SSID1/STA_SSID2 the MQTT STATE message reports first one only, not active one

ChristoSimeonoff, 2018-10-04:
Setting the user_config.h entries thus:
#define STA_SSID1      "ssid1"
#define STA_PASS1     "pass1"
#define STA_SSID1      "ssid2"
#define STA_PASS1     "pass2"
define WIFI_CONFIG_TOOL       WIFI_RETRY

will try both SSIDs.
Id SSID1 fails to connet, and SSID2 succeeds, the Sonoff comes on-line and reports status.
The MQTT STATUS message erroneously contains "SSId": <SSID1> rather than "SSId": <SSID2>

BUG: The active SSID should be used in reporting.

- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ :  Sonoff dual R2
- [ X] _Development/Compiler/Upload tools used_ : Tasmota 6.2.1 Arduino IDE1.8.4 on OpenSuse Tumbleweed
- [X ] _Provide the output of command_``status 0`` :

sonoff/tele/IoTa_9D72CF/STATE {"Time":"2018-10-04T14:49:06","Uptime":"0T00:19:10","Vcc":3.147,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"<redacted>","RSSI":100,"APMac":"<redacted>"}}

Note: The Sonoff was connected to STA_SSID2 correctly, but is reporting STA_SSID1 (which was powered off)

**To Reproduce**
Set the user_config.h entries thus:

#define STA_SSID1      "ssid1"
#define STA_PASS1     "pass1"
#define STA_SSID1      "ssid2"
#define STA_PASS1     "pass2"
define WIFI_CONFIG_TOOL       WIFI_RETRY

ensure AP for ssid1 is powered off, and AP for ssid2 is available

**Expected behavior**
The STATE message should report the current, active ssid

**Screenshots**
N/A

**Additional context**
N/A

ChristoSimeonoff, 2018-10-04:
The line:
The MQTT STATUS message erroneously contains "SSId": rather than "SSId": 
should read:
The MQTT STATUS message erroneously contains "SSId1": rather than "SSId2":

ChristoSimeonoff, 2018-10-04:
After a number of power cycles, the STATE message changes to:
"Wifi":{"AP":2,"SSid":"correct ssid"
It appears the bug is a delay in updating the STATE message pointer to the active SSID

The problem is reproducible by swapping access points and the SSID1 and SSID2 entries
The bug exists only on the first power up.
a second reboot fixes the problem.

ascillato2, 2018-10-04:
Hi,

Sorry, I could not reproduce your issue. It works as expected in my devices.

Where are you seeing the report of the selected AP? because may be that your are seeing in your HomeAutomation software and if you have a retain flag, you are seeing the previous status.

ascillato2, 2018-10-06:
Hi,

Where are you seeing the report of the selected AP?

ascillato2, 2018-10-06:
Hi,

When you have time, please tell us the requested information. Also, Remember that you can find us in the Tasmota Discord Chat.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-06 by ascillato2