arendst/Tasmota #3973: How to store RfRaw data to a rfkey?

haegar33, 2018-10-04:
- [x ] Provide the output of command ``status 0`` :
```
13:33:58 MQT: stat/sonoff/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:33:58 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:20:34","StartupUTC":"2018-10-04T11:13:24","Sleep":100,"BootCount":13,"SaveCount":22,"SaveAddress":"FA000"}}
13:33:58 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:33:58 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Heimnetz",""],"TelePeriod":3000,"SetOption":["00008009","55818000","00000001"]}}
13:33:58 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
13:33:58 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5417","IPAddress":"192.168.178.39","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"60:01:94:A3:15:29","Webserver":2,"WifiConfig":3}}
13:33:58 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.23","MqttPort":1883,"MqttClientMask":"RF_Bridge","MqttClient":"RF_Bridge","MqttUser":"haegar","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:33:58 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 04 11:33:58 2018","Local":"Thu Oct 04 13:33:58 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:32","Sunset":"18:55"}}
13:33:58 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-04T13:33:58"}}
13:33:58 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-04T13:33:58","Uptime":"0T00:20:34","Vcc":3.142,"Wifi":{"AP":1,"SSId":"Heimnetz","RSSI":100,"APMac":"E0:28:6D:ED:6F:38"}}}
```

I finally found a set of RF powerplugs which can be controlled by the Sonoff RFBridge with the portisch Firmware. However I need to send rather lengthy B0 commands like 
`rfraw AAB004.......`

Now I would like to assign this to one of the rfkeys. Can this raw-string be an argument for a key? Normaly they are defined (or learned) with several parameters on sync, low, high.. values?

I tried:
`RFKEY7  4 {"RfRaw":{"Data":"AAB04E0514013....}}`
which does not work?

Jason2866, 2018-10-04:
Not implemented (yet).
Which HA system do you use?
In OpenHab could this be done as example this way (items):
```
Switch Glaskugel        "Glaskugel"             <light> (LR,gLight)     [ "Lighting" ] 
		{ mqtt=">[broker:cmnd/sonoff-8E8329/Backlog:command:ON:RfRaw AAB0210314016703F924180101011001100110010101100110011001010110010101100255; RfRaw 0],
                >[broker:cmnd/sonoff-8E8329/Backlog:command:OFF:RfRaw AAB0210314016A03EA24360101011001100110010101100110011001010110011001010255; RfRaw 0]" }

```

haegar33, 2018-10-04:
For now my only HA is the RF Bridge itself ;) As the main purpose is to switch on/off certain lights according to a schedule  I might try to define time triggered rules which use the rfraw command? But I am afraid the rule buffers might not be large enough?

Jason2866, 2018-10-04:
Worth a try.
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#rule-syntax
3 sets of rules a 511 byte

Jason2866, 2018-10-04:
As you said the rfraw strings needs a lot of space. 
So i am in doubt that rfkey command will be changed to store this.
Maybe the right time to look around for a HA system that fit your needs.
I can recommend OpenHab ;-)

arendst, 2018-10-04:
If the rule space is not enough then they won't be supported.

Jason2866, 2018-10-04:
@haegar33 
Question is answered (RfKey wil not be changend) 
Please close this issue.
Thank you

closed 2018-10-04 by ascillato2

henfri, 2018-12-09:
Hello,

today, there are three rule stores possible. Does each of them have 511 byte?
In that case, one could use three rules with one command each.

Greetings,
Hendrik

ascillato, 2018-12-09:
Hi, sorry, your question is offtopic to this closed thread. Please address general questions to the Tasmota Support Chat.

Rules are 3 slots of 511 characters. So, you can put all the rules you want on each slot up to 511 characters as ON [trigger] DO [command] ENDON ON [trigger] DO [command] ENDON ON.......

sniperxnl, 2020-01-31:
> switch:
>       platform: mqtt
>       name: 'gang'
>       command_topic: 'cmnd/rf_bridge/rfraw'
>       state_topic: 'stat/rf_bridge/rfraw'
>       availability_topic: 'tele/rf_bridge/LWT'
>       payload_available: 'Online'
>       payload_not_available: 'Offline'
>       payload_on: '#AA B0 4E 05 08 0168 0AE6 00D2 04F6 28BE 481828383828382838283828283828383808382828383828283828383828283838082838283828382838283828382838283838283828283838283828283828383828 55'
>       payload_off: '#A B0 4E 05 08 0168 0AF0 00D2 04F6 28FA 481828383828382838283828283828383808382828383828283828383828283838082838283828382838283828382838283838283828283828383828283828383828 55'
>       optimistic: true

Why this is not working?
Tasmota console says this

> 16:09:17 MQT: stat/rf_bridge/RESULT = {“RfRaw”:“ON”}
> 16:09:29 MQT: stat/rf_bridge/RESULT = {“RfRaw”:“ON”}

ionultd, 2020-02-01:
I get the same when using the console and running
cmnd/Sonoff_Bridge2/Backlog RfRaw AAB04A0308026200E613EC00110011010100101101010011001011001100110101001101001011010100110101010101010100110101010101010101010101001100101101001101001011001255; RfRaw 0

ionultd, 2020-02-02:
@sniperxnl Did you find an answer yet as I have almost exactly the same question as you. If I find an answer I will let you know.

Robmak73, 2020-07-31:
remove the spaces in the b0, if your mqtt communication is all ok that should work. Try the command in tasmota console by typing rfraw AAB04E05 BLAH BLAH BLAH. You should see if your bridge is sending the code correctly this way before publishing your command via mqtt.

bachoo786, 2023-02-28:
@Robmak73 @ionultd @sniperxnl I managed to send the commands via mqtt however on Homeassistant after setting the up the MQTT switch, I get the switch as unavailable but before it worked and after playing around and adding another one none of it works on Homeassistant.

Could you guys please share your config? thanks.