arendst/Tasmota #3975: KNX SwitchToggle

nitinuae, 2018-10-04:
Hello

KNX part of Tasmota is missing a few functionalities.
One of the most important is button toggle (Right now its push only)

Eg - A 3 gang switch is attached to GPIO 1,3,9. The switch is ordinary one way switch(what we use to switch on light). 
Presently, if switch is in OFF position, and switched ON, a KNX telegram is sent.
But, if switch is ON, and switched OFF, there is no event capturing by KNX and nothing is sent. Can this event be captured and used by KNX to send a telegram?

I want to put in wemosD1 in every switch box - and want a telegram to be sent (it can be 1 or 0) everytime the switch changes state. This way, wemos will send a telegram to my GIRA and I can write a logic to toggle the light. I will not have to change the existing one way switches. 

It doesnot matter if the switch is looking down(what we call ON) or looking up(what we call OFF). As long as state is changed, it should send the telegram.

Thank you
nitin

ascillato, 2018-10-04:
Hi,

From the KNX webmenu it only supports buttons (push buttons) by now.

For switches you can set in KNX_TX1 the address you want to send the command and use the following rule to be typed in the Tasmota Console:

```
rule1 1
rule1 on switch1#state do KnxTx_Cmnd1 %value% endon
```

nitinuae, 2018-10-04:
Perfect - Thank you Ascillato.
For multiple switches, I use backlog?

closed 2018-10-04 by ascillato2