arendst/Tasmota #3977: Learning RF code for Cotech RF sockets

b1g1an, 2018-10-04:
I've flashed my Sonoff RF Bridge with 6.2.1, changed SetOption28 to 1 and set up some Status rf sockets using the code shown in the console to work with Home Assistant, all good so far.

I then tried to add some Cotech sockets, seen here... 

https://www.instructables.com/id/Control-CoTech-Remote-Switch-With-Arduino-433Mhz/

These produce a different seven digit code nearly every time and none work when sending. They worked fine with an RMPro that eventually died but that used codes in this format...

sgGEAw8kHxMgEw4kIBMgEiATHxQOJCATDiQfFB8TDyQOJR8THxMgEx8TIBMPJB8TDyQfE2PuDyQgEx8TDyQfEyATHxMgEw4kIBMPIyATIBIPJA8jIBMfFB8TIBMfEw8kHxMPJB8YC00LJyQPIw8LKCMPCikKKAooJA8jDyQOCygjDwooCygKKSMPIw8kDgsoCigkDwooJA8KTQsoIw8kDgsoJA4LKAooCikjDyQOJA8KKCQPCigLKAooIw8kDyMPCigLKCQOCygjDwtNCigkDyMPCygjDwooCygKKCMQIw8jDwsoIw8LKAooCigkDyMPJA4LKAooJA8KKCMPC04KKCQOJA8KKCQOCygKKAsnJA8kDiQPCigkDgsoCigLKCMPJA8jDwooCygjDwsnJA9j7g8kHxMgEw8jIBMPJA4kDyMgEyATHxQOJB8UDiQOJA8kIBIgEyASDyQPJB8UDiQfE2PvDiQgEx8TDyQfEw8kDyMPJCASIBMfFA4kHxMPJA8jECMgEx8TIBMPIw8kIBIPJB8UYu8PIyATHxQOJCASDyQPIw8kHxMgEyATDyMgEw8jDyQPIyATHxMgEw8kDiQfFA4kIBNj7g8kHxMgEw4lHxMPIw8kDyMgEyATHxMPJB8TDyQOJA8kHxMgEyASDyQPIyATDyQfGApOCigkDyMPCigkDwooJA8jDyMPCygjDwopCigjDwsoIw8LKAooCigLKAooJA4LKCQOC00LKCMPJA4LKCMQCigkDiQOJA8LKCMPCigKKCQPCyckDwooCygKKAsnCyclDgopIw8LTQsnJA8jDwsoIw8KKSMPJA4kDwooJA8KKAooJA8KKCQOCygKKAsoCigLJyQPCigkDwpOCigkDyMPCigkDwooJA8jDyMPCygjDwooCygkDgsoJA4LKAooCigLJwsoJA4LKCMQYu8PIyATHxMPJCASDyQgEiATIBMOJCATDiUOJB8TDyQgEg8kDyMPJA8kDiQgEw4kIBNj7g8kHxMgEw8jIBMPIyATHxMgEw8kHxMPJA8jIBMPIyATDyMPJA8kDiQPIyATDyQfE2PvDyQfEx8TDyQfEw8kIBIgEyATDiQgEw8jDyQgEg8kIBMOJA8kDiQPJA4kIBMPIyATY+4PJB8TIBMPJB8TDyQfEyATHxMPJB8UDiQPIyATDiUfEw8kDiQPJA4kDyQfEw8kIAAF3AAAAAA=

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff RF Bridge
- [X] _Development/Compiler/Upload tools used_ : ESP Easy
- [X] _Provide the output of command_``status 0`` :
```
18:22:06 MQT: stat/RF_Bridge/STATUS = {"Status":{"Module":25,"FriendlyName":["RF_Bridge"],"Topic":"RF_Bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
18:22:06 MQT: stat/RF_Bridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"1T01:05:18","StartupUTC":"2018-10-03T16:16:48","Sleep":0,"BootCount":8,"SaveCount":14,"SaveAddress":"F6000"}}
18:22:06 MQT: stat/RF_Bridge/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:22:06 MQT: stat/RF_Bridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["secret",""],"TelePeriod":300,"SetOption":["10008029","55818000","00000000"]}}
18:22:06 MQT: stat/RF_Bridge/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
18:22:06 MQT: stat/RF_Bridge/STATUS5 = {"StatusNET":{"Hostname":"RF_Bridge","IPAddress":"secret","Gateway":"secret","Subnetmask":"255.255.255.0","DNSServer":"secret","Mac":"secret","Webserver":2,"WifiConfig":5}}
18:22:06 MQT: stat/RF_Bridge/STATUS6 = {"StatusMQT":{"MqttHost":"secret","MqttPort":1883,"MqttClientMask":"RF_Bridge","MqttClient":"RF_Bridge","MqttUser":"secret","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:22:06 MQT: stat/RF_Bridge/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 04 17:22:06 2018","Local":"Thu Oct 04 18:22:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:54","Sunset":"18:23"}}
18:22:06 MQT: stat/RF_Bridge/STATUS10 = {"StatusSNS":{"Time":"2018-10-04T18:22:06"}}
18:22:06 MQT: stat/RF_Bridge/STATUS11 = {"StatusSTS":{"Time":"2018-10-04T18:22:06","Uptime":"1T01:05:18","Vcc":3.182,"Wifi":{"AP":1,"SSId":"secret","RSSI":100,"APMac":"secret"}}}
18:22:38 MQT: tele/RF_Bridge/STATE = {"Time":"2018-10-04T18:22:37","Uptime":"1T01:05:49","Vcc":3.182,"Wifi":{"AP":1,"SSId":"secret","RSSI":100,"APMac":"secret"}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
Press button on remote, resend code seen in console

**Expected behavior**
_A clear and concise description of what you expected to happen._
Code switches on/off socket

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-10-04:
Flash the bridge with custom firmware for the EFM MCU
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433
For further questions meet us as Discor https://discord.gg/Ks2Kzd4

ascillato2, 2018-10-04:
Hi,

Changing codes are not supported. Sorry.

andrethomas, 2018-10-04:
That code is an MD5 of something... the question is what is the something, so it's probably not going to happen unless you know what is inside the MD5 - May also be AES->MD5

Jason2866, 2018-10-04:
The Cotech sockets should work with Portisch EFM Firmware.
The code you sniffed isnt from this sockets. As @andrethomas wrote this devices wont work.

andrethomas, 2018-10-04:
Correction, I meant Base64 encoded not MD5 - Sorry had a long day.

Anyway, to get back to it - if you decode the base64 packet you posted above it appears to contain binary data which may be useful.

andrethomas, 2018-10-04:
So that base64 packet decodes to this in hex

`b2 01 84 03 0f 24 1f 13 20 13 0e 24 20 13 20 12 20 13 1f 14 0e 24 20 13 0e 24 1f 14 1f 13 0f 24 0e 25 1f 13 1f 13 20 13 1f 13 20 13 0f 24 1f 13 0f 24 1f 13 63 ee 0f 24 20 13 1f 13 0f 24 1f 13 20 13 1f 13 20 13 0e 24 20 13 0f 23 20 13 20 12 0f 24 0f 23 20 13 1f 14 1f 13 20 13 1f 13 0f 24 1f 13 0f 24 1f 18 0b 4d 0b 27 24 0f 23 0f 0b 28 23 0f 0a 29 0a 28 0a 28 24 0f 23 0f 24 0e 0b 28 23 0f 0a 28 0b 28 0a 29 23 0f 23 0f 24 0e 0b 28 0a 28 24 0f 0a 28 24 0f 0a 4d 0b 28 23 0f 24 0e 0b 28 24 0e 0b 28 0a 28 0a 29 23 0f 24 0e 24 0f 0a 28 24 0f 0a 28 0b 28 0a 28 23 0f 24 0f 23 0f 0a 28 0b 28 24 0e 0b 28 23 0f 0b 4d 0a 28 24 0f 23 0f 0b 28 23 0f 0a 28 0b 28 0a 28 23 10 23 0f 23 0f 0b 28 23 0f 0b 28 0a 28 0a 28 24 0f 23 0f 24 0e 0b 28 0a 28 24 0f 0a 28 23 0f 0b 4e 0a 28 24 0e 24 0f 0a 28 24 0e 0b 28 0a 28 0b 27 24 0f 24 0e 24 0f 0a 28 24 0e 0b 28 0a 28 0b 28 23 0f 24 0f 23 0f 0a 28 0b 28 23 0f 0b 27 24 0f 63 ee 0f 24 1f 13 20 13 0f 23 20 13 0f 24 0e 24 0f 23 20 13 20 13 1f 14 0e 24 1f 14 0e 24 0e 24 0f 24 20 12 20 13 20 12 0f 24 0f 24 1f 14 0e 24 1f 13 63 ef 0e 24 20 13 1f 13 0f 24 1f 13 0f 24 0f 23 0f 24 20 12 20 13 1f 14 0e 24 1f 13 0f 24 0f 23 10 23 20 13 1f 13 20 13 0f 23 0f 24 20 12 0f 24 1f 14 62 ef 0f 23 20 13 1f 14 0e 24 20 12 0f 24 0f 23 0f 24 1f 13 20 13 20 13 0f 23 20 13 0f 23 0f 24 0f 23 20 13 1f 13 20 13 0f 24 0e 24 1f 14 0e 24 20 13 63 ee 0f 24 1f 13 20 13 0e 25 1f 13 0f 23 0f 24 0f 23 20 13 20 13 1f 13 0f 24 1f 13 0f 24 0e 24 0f 24 1f 13 20 13 20 12 0f 24 0f 23 20 13 0f 24 1f 18 0a 4e 0a 28 24 0f 23 0f 0a 28 24 0f 0a 28 24 0f 23 0f 23 0f 0b 28 23 0f 0a 29 0a 28 23 0f 0b 28 23 0f 0b 28 0a 28 0a 28 0b 28 0a 28 24 0e 0b 28 24 0e 0b 4d 0b 28 23 0f 24 0e 0b 28 23 10 0a 28 24 0e 24 0e 24 0f 0b 28 23 0f 0a 28 0a 28 24 0f 0b 27 24 0f 0a 28 0b 28 0a 28 0b 27 0b 27 25 0e 0a 29 23 0f 0b 4d 0b 27 24 0f 23 0f 0b 28 23 0f 0a 29 23 0f 24 0e 24 0f 0a 28 24 0f 0a 28 0a 28 24 0f 0a 28 24 0e 0b 28 0a 28 0b 28 0a 28 0b 27 24 0f 0a 28 24 0f 0a 4e 0a 28 24 0f 23 0f 0a 28 24 0f 0a 28 24 0f 23 0f 23 0f 0b 28 23 0f 0a 28 0b 28 24 0e 0b 28 24 0e 0b 28 0a 28 0a 28 0b 27 0b 28 24 0e 0b 28 23 10 62 ef 0f 23 20 13 1f 13 0f 24 20 12 0f 24 20 12 20 13 20 13 0e 24 20 13 0e 25 0e 24 1f 13 0f 24 20 12 0f 24 0f 23 0f 24 0f 24 0e 24 20 13 0e 24 20 13 63 ee 0f 24 1f 13 20 13 0f 23 20 13 0f 23 20 13 1f 13 20 13 0f 24 1f 13 0f 24 0f 23 20 13 0f 23 20 13 0f 23 0f 24 0f 24 0e 24 0f 23 20 13 0f 24 1f 13 63 ef 0f 24 1f 13 1f 13 0f 24 1f 13 0f 24 20 12 20 13 20 13 0e 24 20 13 0f 23 0f 24 20 12 0f 24 20 13 0e 24 0f 24 0e 24 0f 24 0e 24 20 13 0f 23 20 13 63 ee 0f 24 1f 13 20 13 0f 24 1f 13 0f 24 1f 13 20 13 1f 13 0f 24 1f 14 0e 24 0f 23 20 13 0e 25 1f 13 0f 24 0e 24 0f 24 0e 24 0f 24 1f 13 0f 24 20 00 05 dc 00 00 00 00`

andrethomas, 2018-10-04:
I think given this, it will not be possible to handle directly inside Tasmota. My recommendation is that you post at https://github.com/Portisch/RF-Bridge-EFM8BB1 to see if they can assist with the necessary decoding so that if it can be decoded then it may be useful in Tasmota.

So for now I do not think it is a Tasmota issue but rather one that should be referred to https://github.com/Portisch/RF-Bridge-EFM8BB1 for further discussion there.

andrethomas2, 2018-10-04:
Closing this issue for now as it is not related directly to Tasmota firmware and you are referred to https://github.com/Portisch/RF-Bridge-EFM8BB1 to explore this further. Once you get something back that is implementable in Tasmota you are welcome to open an issue again if there is something else to do in Tasmota. The latter is unlikely as if support is added to the Portish firmware then Tasmota should be able to handle it without any further modification.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-04 by andrethomas2

b1g1an, 2018-10-05:
Thanks for all the info guys, I'll mod and flash the new RF firmware onto the bridge tomorrow, anyone seen an idiots guide to using the new features it brings, the existing is more this is all the commands and relying on you knowing what to do with them rather than just do this and it'll do what you want.