arendst/Tasmota #3978: Tasmota Front end reachable 50% of the time

fiservedpi, 2018-10-04:
When visiting the local IP of the tasmota I'm met with a connection time out message in chrome. After 3 or 4 refreshes the front end shows up, if I click an option such as MQTT config same thing have to refresh 4-5 times for it show

reloxx13, 2018-10-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.
a
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-04:
Hi,

Sorry. can't reproduce.

Please, complete the troubleshooting template so as to have the information for properly help you.

closed 2018-10-04 by ascillato2