arendst/Tasmota #3980: Tasmota 6.2.1 Non-functional on Sonoff Slampher

SalSuleman, 2018-10-05:
**Describe the bug**
When OTA updated or freshly flashed from 5.8.0h to 6.2.1 the device becomes non functional and blinks the led in 1 second intervals

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : Sonoff Slampher 
- [X ] _Development/Compiler/Upload tools used_ : Arduino 1.8.5 and Platformio
- [ ] _Provide the output of command_``status 0`` : N/A

**To Reproduce**
 OTA update or Flash with arduino ide or platformio  from 5.8.0h to 6.2.1

**Expected behavior**
_A clear and concise description of what you expected to happen._
normal function of the device

**Additional context**
_Add any other context about the problem here._
Might be due to the fact that slampher is ESP8285 based

andrethomas, 2018-10-05:
Try esptool erase flash, and re-flash with released binary and report back if anything is different.

ascillato2, 2018-10-06:
Hi,

When you have time, please tell us if this recommendations have work for you. If not, remember that you can find us in the Tasmota Discord Chat.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-06 by ascillato2

SalSuleman, 2018-10-07:
Re-flashing ended up with the same results