arendst/Tasmota #3981: [Sonoff-Tasmota 6.2.1] - Sonoff 4ch

danielraq, 2018-10-05:
Hi,
I'm trying to add additional sensor to my sonoff 4ch on grio3 (regular pin is already occupied) , i've disabled serial loggin by `SerialLog off`, unfortunately it didn't help. Could you help me and tell me if it's possible, if yes what i'm doing wrong.

Jason2866, 2018-10-05:
A little more Info would be helpful :-)
What sensor? Is it correct wired? Is in firmware sensor enabled on GPIO3?

andrethomas, 2018-10-05:
I would aim for D1 (GPIO4) - It is also broken out nicely on the the board

[image]

Jason2866, 2018-10-05:
@andrethomas 
if it is a 4CHpro yes. On a 4CH GPIO 2 is easy reachable

andrethomas, 2018-10-05:
@Jason2866 Aaah, even easier.. so why is there this issue then?

danielraq, 2018-10-05:
@Jason2866 

> What sensor? 
Ir reciver - TSOP4838 

> Is it correct wired? 
Yes, it works with GPIO2.

> Is in firmware sensor enabled?  Configured on GPIO3?
I'm not sure if I understood the question so correct me if I'm wrong. It is enabled in tasmota `51 IRrecv`

@andrethomas
I'm using regular 4CH sonoff.

To be precise GPIO2 is already in use by DHT11 sensor, This is why i decided to use GPIO3

andrethomas, 2018-10-05:
@danielraq 

I do not see that GPIO3 is broken out on the module so not sure where you connected it to.

[image]

Jason2866, 2018-10-05:
Finally you can solder on Module GPIO16
https://www.itead.cc/wiki/images/thumb/e/e6/PSF-A85_pinmap.jpg/500px-PSF-A85_pinmap.jpg

danielraq, 2018-10-05:
@Jason2866
it worked. i switched cables and it's working :) thanks

closed 2018-10-05 by danielraq