arendst/Tasmota #3982: Problems Sonoff Basic R2 V1.0 2017-01-11

raymajor, 2018-10-05:
Problems with Sonoff Basic R2 V1.0 2017-01-11

See:
https://github.com/arendst/Sonoff-Tasmota/issues/2868#issuecomment-427127450
https://github.com/arendst/Sonoff-Tasmota/issues/2868#issuecomment-427326953

V5.14 works fine.
Update to V6.1.1 minimum 
Update to V6.1.1 Sonoff basic wil reboot.

00:00:15 CMD: status 0
00:00:15 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:15 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:14","StartupUTC":"","Sleep":0,"BootCount":28,"SaveCount":41,"SaveAddress":"FA000"}}
00:00:15 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-23T11:22:18","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:15 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RayMan2.4",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:00:15 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000413","0FDAE794","0C000000","23B6179E","00000000"]}}
00:00:15 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0769","IPAddress":"0.0.0.0","Gateway":"192.168.10.254","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"BC:DD:C2:0F:43:01","Webserver":2,"WifiConfig":1}}
00:00:15 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.75","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F4301","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:15 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:15 1970","Local":"Thu Jan 01 00:00:15 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:15 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:15"}}
00:00:15 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:15","Uptime":"0T00:00:14","Vcc":3.196,"POWER":"OFF","Wifi":{"AP":1,"SSId":"RayMan2.4","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}

Jason2866, 2018-10-05:
Erase device with esptool (described in wiki). No upgrade via OTA! Just use latest firmware version. This link: https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.2.1/sonoff.bin
This is the only way to get a defined state.

andrethomas, 2018-10-05:
I see your wificonfig = 1 on that status 0 output- perhaps consider WifiConfig 2 and once everything is working consider WifiConfig 5

Check wiki here https://github.com/arendst/Sonoff-Tasmota/wiki/Commands to see what the different wificonfig commands do to the way wifi behaves.

andrethomas, 2018-10-05:
Once you've completed @Jason2866 recommendation and you get console access, please give us the output of status 0 again. Thanks

ascillato2, 2018-10-06:
Hi,

When you have time, please tell us if this recommendations have work for you. If not, remember that you can find us in the Tasmota Discord Chat.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-06 by ascillato2

raymajor, 2018-10-06:
Sorry Guy's was a busy day...

I have delete flash with: esptool.exe -cp COM4 -ce
Than upload the sonoff.bin like @Jason2866 said.
That goes fine.. Sonoff R2 goes into WIFI manager. But i can't connect to it with the iPhone.

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0F4301, GroupTopic sonoffs) Version 6.2.1.11-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-0769 with IP address 192.168.244.1

Why the webver is not ip 192.168.4.1?

Below the latest Staus 0:
status 0
00:02:33 CMD: status 0
00:02:33 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:02:33 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:33","StartupUTC":"","Sleep":0,"BootCount":29,"SaveCount":30,"SaveAddress":"F6000"}}
00:02:33 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.11","BuildDateTime":"2018-10-04T16:00:46","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:02:33 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:02:33 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":479,"Free":524,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","000003C0"]}}
00:02:33 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0769","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"BC:DD:C2:0F:43:01","Webserver":2,"WifiConfig":5}}
00:02:33 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F4301","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:02:33 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:02:33 1970","Local":"Thu Jan 01 00:02:33 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:02:33 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:02:33"}}
00:02:33 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:02:33","Uptime":"0T00:02:33","Vcc":3.160,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100}}}

Now i'am trying the Wif options like @andrethomas said.

raymajor, 2018-10-06:
Starting Wificonfig 2:

wificonfig 2
00:00:53 CMD: wificonfig 2
00:00:53 RSL: stat/sonoff/RESULT = {"WifiConfig":"WifiManager selected"}
00:00:55 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0F4301, GroupTopic sonoffs) Version 6.2.1.11-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-0769 with IP address 192.168.244.1

Connect iPhone WIFI to Sonoff-0769
No connection with netwerk 'Sonoff-0769"

Status 0:
Status 0
00:01:56 CMD: Status 0
00:01:56 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:01:56 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:56","StartupUTC":"","Sleep":0,"BootCount":33,"SaveCount":35,"SaveAddress":"F9000"}}
00:01:56 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.11","BuildDateTime":"2018-10-04T16:00:46","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:01:56 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:01:56 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":479,"Free":524,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","000003C0"]}}
00:01:56 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0769","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"BC:DD:C2:0F:43:01","Webserver":2,"WifiConfig":2}}
00:01:56 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F4301","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:56 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:56 1970","Local":"Thu Jan 01 00:01:56 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:56 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:56"}}
00:01:56 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:56","Uptime":"0T00:01:56","Vcc":3.160,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100}}}
00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:01"}

raymajor, 2018-10-06:
backlog ssid1 RayMan2.4;password1 xxxxx;
00:01:24 CMD: backlog ssid1 RayMan2.4;password1 xxxxx;
00:01:24 RSL: stat/sonoff/RESULT = {"SSId1":"RayMan2.4"}
00:01:24 RSL: stat/sonoff/RESULT = {"Password1":"xxxxxx"}
00:01:26 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0F4301, GroupTopic sonoffs) Version 6.2.1.11-2_3_0
00:00:00 WIF: Connecting to AP1 RayMan2.4 in mode 11N as sonoff-0769...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-0769.local with IP address 192.168.10.99
00:00:05 MQT: Attempting connection...
00:00:06 DNS: Query done. MQTT services found 0
00:00:06 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:32:33 MQT: Attempting connection...
19:32:34 DNS: Query done. MQTT services found 0
19:32:34 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:32:43 WIF: Connecting to AP1 RayMan2.4 in mode 11N as sonoff-0769...
19:32:50 WIF: Connect failed as AP cannot be reached
19:32:50 WIF: Connecting to AP1 RayMan2.4 in mode 11N as sonoff-0769...
19:32:57 WIF: Connect failed as AP cannot be reached
19:32:57 WIF: WifiManager active for 3 minutes
19:32:57 HTP: Web server stopped
19:32:57 HTP: Web server active on sonoff-0769 with IP address 192.168.4.1

I was in to it on Ip 192.168.10.99 threu web. Not on the iPhone.
It was very slow..
Than i get no more connections.
LED is still flashing..

status 0
19:38:23 CMD: status 0
19:38:23 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:38:23 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:29","StartupUTC":"2018-10-06T18:34:54","Sleep":0,"BootCount":5,"SaveCount":9,"SaveAddress":"FB000"}}
19:38:23 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.11","BuildDateTime":"2018-10-04T16:00:46","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:38:23 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RayMan2.4",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
19:38:23 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":479,"Free":524,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","000003C0"]}}
19:38:23 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0769","IPAddress":"0.0.0.0","Gateway":"192.168.10.254","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"BC:DD:C2:0F:43:01","Webserver":2,"WifiConfig":2}}
19:38:23 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F4301","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:38:23 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 06 18:38:23 2018","Local":"Sat Oct 06 19:38:23 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:57","Sunset":"18:19"}}
19:38:23 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-06T19:38:23"}}
19:38:23 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-06T19:38:23","Uptime":"0T00:03:29","Vcc":3.202,"POWER":"OFF","Wifi":{"AP":1,"SSId":"RayMan2.4","BSSId":"70:4F:57:8A:AD:03","Channel":1,"RSSI":100}}}

raymajor, 2018-10-06:
What i also see is that this version is V6.2.1.11 on the R2
On al the other Basis R1 devices it's 6.2.1

It's now again @ Wifi manager

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0F4301, GroupTopic sonoffs) Version 6.2.1.11-2_3_0
00:00:00 WIF: Connecting to AP1 RayMan2.4 in mode 11N as sonoff-0769...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-0769.local with IP address 192.168.10.99
00:00:05 MQT: Attempting connection...
00:00:06 DNS: Query done. MQTT services found 0
00:00:06 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:40:18 MQT: Attempting connection...
19:40:19 DNS: Query done. MQTT services found 0
19:40:19 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:40:30 MQT: Attempting connection...
19:40:31 DNS: Query done. MQTT services found 0
19:40:31 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:40:42 MQT: Attempting connection...
19:40:43 DNS: Query done. MQTT services found 0
19:40:43 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:40:53 MQT: Attempting connection...
19:40:54 DNS: Query done. MQTT services found 0
19:40:54 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:41:05 MQT: Attempting connection...
19:41:06 DNS: Query done. MQTT services found 0
19:41:06 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:41:17 MQT: Attempting connection...
19:41:18 DNS: Query done. MQTT services found 0
19:41:18 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:41:29 MQT: Attempting connection...
19:41:30 DNS: Query done. MQTT services found 0
19:41:30 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:41:40 MQT: Attempting connection...
19:41:41 DNS: Query done. MQTT services found 0
19:41:41 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:41:52 MQT: Attempting connection...
19:41:53 DNS: Query done. MQTT services found 0
19:41:53 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:42:04 MQT: Attempting connection...
19:42:05 DNS: Query done. MQTT services found 0
19:42:05 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
19:42:15 WIF: Connect failed as AP cannot be reached
19:42:15 WIF: Connecting to AP1 RayMan2.4 in mode 11N as sonoff-0769...
19:42:23 WIF: Connect failed as AP cannot be reached
19:42:23 WIF: WifiManager active for 3 minutes
19:42:23 HTP: Web server stopped
19:42:23 HTP: Web server active on sonoff-0769 with IP address 192.168.4.1

But i can not connect to it on a iPhone....

Why is this happend? i"ll have al the Sonoff stuf in my House. No problems like this with flashing to Tasmota....

andrethomas, 2018-10-06:
The 6.2.1 devices you flashed from the released binaries.
The 6.2.1.11 device is flashed with the current development codebase binaries.

The sonoff basic R2 is problematic in some limited reported cases - The theory is that the power supply is not strong enough or very much borderline.

Some people have managed to solve this by adding a seperate 3.3V during the flash process but there is no scientific reason other than limitations of a specific usb serial device that can explain this.

raymajor, 2018-10-06:
Are you looking for a solution for the Sonoff basic R2?

Jason2866, 2018-10-06:
@raymajor 
If it is a hardware issue there is no software solution.
You can stay on a older Tasmota Version which does not need so much (CPU) power.

andrethomas, 2018-10-06:
Perhaps, if you have the time and inclination, humour me on testing this binary on the R2

http://thehackbox.org/sonoff-basic_R2.bin

It is basically a compiled binary with the minimal stuff enabled to work on a Sonoff basic R2.

Let us know if it behaves any different to the standard releases.

raymajor, 2018-10-06:
i'm gonne try @andrethomas 
Don't close this topic... ;-)

raymajor, 2018-10-06:
@andrethomas This one works...

Status 0
21:19:17 CMD: Status 0
21:19:17 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:19:17 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"","RestartReason":"Power on","Uptime":"0T00:00:29","StartupUTC":"2018-10-06T20:18:48","Sleep":0,"BootCount":5,"SaveCount":11,"SaveAddress":"F9000"}}
21:19:17 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.11","BuildDateTime":"2018-10-06T21:49:17","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:19:17 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RayMan2.4",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
21:19:17 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":418,"Free":584,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F000680","00000000","00000096","000000C0"]}}
21:19:17 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5219","IPAddress":"192.168.10.64","Gateway":"192.168.10.254","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"80:7D:3A:45:D4:63","Webserver":2,"WifiConfig":2}}
21:19:17 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.75","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_45D463","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:19:17 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 06 20:19:17 2018","Local":"Sat Oct 06 21:19:17 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:57","Sunset":"18:19"}}
21:19:17 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-06T21:19:17"}}
21:19:17 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-06T21:19:17","Uptime":"0T00:00:29","Vcc":3.252,"POWER":"OFF","Wifi":{"AP":1,"SSId":"RayMan2.4","BSSId":"70:4F:57:8A:AD:03","Channel":11,"RSSI":82}}}

andrethomas, 2018-10-06:
@arendst @ascillato Perhaps note the outcome of this issue.

Release binary from 6.2.1 gave the behaviour we've seen on 2 or so sonoff basic R2's

With the theory that the power supply may not be having enough current, I considered a minimalistic compilation with no support for I2C devices as to decrease the CPU demand for power during startup when all the detection of devices is taking place.

User @raymajor claims the compiled binary from using the attached user_config.h has resulted in a working sonoff basic R2 which was previously not working as expected.

So its either a fluke that this specific flash instance worked on the R2 or there is some logic to my madness.
[user_config_basic_R2.zip](https://github.com/arendst/Sonoff-Tasmota/files/2453387/user_config_basic_R2.zip)

raymajor, 2018-10-06:
@andrethomas 

I'll flash it @ 74880 baud and @ 115200 baud. Both are working.
I used Backlog ssid1 & @ iPhone log in. That also works. But ipadress on iPhone was 192.168.244.1

Only thing i'll see is that the LED is flashing when there is no MQTT adress is giving in. But may be that's normal...

Jason2866, 2018-10-06:
@raymajor 
You could help us a lot. 
Your Basic R2 seems to be one of the seldom samples which fails with latest Tasmota.
We dont have such a "special" device. Could you send us this for further testing?
You get this back or if we kill it a other brand new Sonoff Basic back
Where do you are from? If from Europe you can send it to me (i am from Germany)
Thank you!
You could find us at Discord Chat

arendst, 2018-10-06:
Now on to the offending I2C device...

raymajor, 2018-10-06:
@Jason2866 I can send it to @arendst Where dutch guys ;-) May be that's the short way. I'll travel every day from im Holland...
I'll have two Basic R2 with the same problems..

andrethomas, 2018-10-06:
For now what I've done is added a sonoff-basic_R2.bin version on the pre-compiled development binaries at http://thehackbox.org/tasmota/ so when we find another R2 we can also refer there and get more confirmation if it works or not.

andrethomas, 2018-11-11:
@raymajor 

Hi,

Since you have a Sonoff Basic R2 - There is a change we would like you to test to see if it solves the problems you previously experienced with the sonoff.bin firmware.

Kindly let us know if you are able to perform a test without causing too much problems with your current setup in case you need to revert to sonoff-basic.bin

Thanks

raymajor, 2018-11-11:
@andrethomas I'll will. 
The R2 is now running on 6.3.0 Dutch ands works fine.

raymajor, 2018-11-11:
@andrethomas 
Hi, just tested with V6.3.0.7 
First OTA. That went wrong.
Second time Seriel. And that's works fine.
I'll think mutch beter than V6.3.0

andrethomas, 2018-11-11:
@raymajor Thanks

@arendst 

User reported 6.3.0 worked fine on his R2 except that it will not always connect first time around and sometimes go into wifi manager mode before eventually connecting.

With the 2a2b224 development binary it seems to connect to wifi on first attempt.

The fact that OTA failed on first attempt is a little concerning, but it may be because of the 6.3.0 firmware, not clear.

The latter is as a result of discussion on discord chat.

FutureCow, 2018-11-11:
I also have the Basic R2

I updated to version 6.3.0 via OTA. This worked, but after a restart there where some problems. The Webinterface was slow and also controlling the light took a second or 2(normally this was instant).

So i flashed 6.3.0-Basic, this was even more problematic! Results of a ping to the R2 was around 98% time out. 
So i connected a laptop to the R2 to flash 6.3.0-minimal. After that I flashed 6.3.0.7 (normal, not basic)
That's running for 10 minutes now without a problem. Will report back if I found a problem  with 6.3.0.7.

andrethomas, 2018-11-11:
@FutureCow That is good news indeed. Thanks for the update.

umetnic, 2018-11-11:
> So i connected a laptop to the R2 to flash 6.3.0-minimal. After that I flashed 6.3.0.7 (normal, not basic)
> That's running for 10 minutes now without a problem. Will report back if I found a problem with 6.3.0.7.

Did the same with my Basic R2, which was also really slow with 6.3.0.
On 6.3.0.7(2a2b224-sonoff) UI works normally again!

andrethomas, 2018-11-11:
@umetnic Thanks for your feedback - much appreciated!