arendst/Tasmota #3983: Help required on flashing RF bridge with EFM8BB1

vaidyasr, 2018-10-05:
Could some one please guide me how to flash RF bridge with EFM8BB1 after flashing Tasmota?. My Sonoff bridge says V2 and not R2. From the Wiki, I can't understand. Can I directly flash the .hex firmware via webpage?. or do I need to connect the cables GPIO4 and C2Ck and GPIO5 and C2D.

andrethomas, 2018-10-05:
>  Can I directly flash the .hex firmware via webpage?

Yes

> do I need to connect the cables GPIO4 and C2Ck and GPIO5 and C2D

Yes, just like it is in the picture and do not power from USB - Power via 3.3V and GND from your USB Adapter - do not connect TX/RX pins - only 3V3 and GND.

It will update the hex and beep when its done.

closed 2018-10-05 by ascillato2

vaidyasr, 2018-10-06:
I am always getting "Failed to init RF chip" error. What am I doing wrong?

vaidyasr, 2018-10-06:
OK I fixed it. The picture in the Wiki is wrong. I followed the picture instead of connection description. The picture shows connection GPIO4 with C2D and GPIO5 with C2CK which is vice versa.

vaidyasr, 2018-10-06:
I am not getting any results on 'rfraw 255' which is supposed to print the firmware version? The output is 
```
06:09:42 CMD: rfraw 255
06:09:42 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}
06:09:42 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA0055"}}
```

vaidyasr, 2018-10-09:
@andrethomas @ascillato2  help please

andrethomas, 2018-10-09:
I also get

```
18:19:22 CMD: rfraw 255
18:19:22 MQT: stat/bridge1/RESULT = {"RfRaw":"ON"}
18:19:22 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA0055"}}
```

What did you expect?

Remember to do Rfraw 0 when you're done.