arendst/Tasmota #3985: Wrong wifi password on sonoff basic

Pierremenard, 2018-10-05:
Hello, I installed Tasmota 6.2.1 on 4 Sonoff 4ch Pro R2 without any problem. Then I tried with a Sonoff Basic R2.
On the Basic the serial output is keeping saying that the wifi password is wrong, but I'm pretty sure it's not.
I re-configured thorugh wifi manager, but still wrong password. I erased the flash and tried with 6.1.1 but still the serial output says wrong password.
I tried also with smartconfig, but it fails to connect.
I have a Asus RT-AC68U router and a dsl-ac52u. I disabled the 5 ghz radio as per wiki.
Any suggestion?

Thank you

Francesco

ascillato, 2018-10-06:
Please, try to erase all flash with esptool.py and then try to flash the precompiled bins from http://thehackbox.org/tasmota/

mrloddo, 2018-10-06:
Hi all,
I think to have the same issue. I've tried several time to use compiled Tasmota 6.2.1 and a couple of previous releases. I've never been able to use Wifi, it continues to retry connections. I've tried with my router and two different mobile hotspot. If I use prebuild release (6.2.1), the wifi works fine.
But I need to change a couple of parameters (principally MQTT username & password size). 
Do you know where I'm doing wrong?

Thanks
Marco

Pierremenard, 2018-10-06:
I installed Tasmota V5 and everything works. I don't dare to upgrade...

Thank you anyway!

Francesco

ascillato2, 2018-10-06:
> I installed Tasmota V5 and everything works. I don't dare to upgrade...

Ok, no problem at all.

If you want to retake this issue, please, just ask to reopen. Thanks.

closed 2018-10-06 by ascillato2

Jason2866, 2018-10-06:
Please try this version if it solves your issue
http://thehackbox.org/tasmota/sonoff-basic_R2.bin
Please report here if it works or not.

andrethomas, 2018-11-11:
@Pierremenard 

Hi

Is your device available to test a new binary which might solve the problem for Sonoff Basic R2's on the 6.3.x firmware?

If so, please let us know.

Thanks

Pierremenard, 2018-11-12:
@andrethomas 
I've already upgraded to 6.2.1 OTA without problems, the problem was just with flashed firmware via serial. I'll update to 6.3 and I'll let you know, but I don't expect any problem. Unfortunately I don't have any spare R2 to check with serial upload and I don't want to risk the production environment.

andrethomas, 2018-11-12:
@Pierremenard Thanks, no prob - There was a fix done which seems to have solved the wifi problem for sonoff basic modules but I agree on not risking a device that is in use for test purposes.