arendst/Tasmota #3987: Separate SetOption30 on 4CH

gnolnos, 2018-10-06:
I have 3 installed Tasmota on 4CHs. But for each connector, I have difference kinds of devices. Ex. Relay 1,2 turn on/off some lights, Relay 3 for room ventilator, Relay 4 for something else. In home-assistant, I have MQTT discovery actived, and all 4CHs detected as switches. I want to set separate kind of channel as light, light, switch, switch, but how to set separate SetOption30?

ascillato, 2018-10-06:
Hi, sorry. It is not supported. You need to configure your relays manually in Home Assistant.

Please, take a look to the wiki. There are several examples for that. 

Thanks

closed 2018-10-06 by ascillato2

gnolnos, 2018-10-06:
Thanks! So manually config each item is the only option.