arendst/Tasmota #3988: WiFi connection status light turns off when switch is on - not solved after 4 weeks ?

L00000813, 2018-10-06:
The following problem still exists, therefore I have opened this new issue (directing it at the previous).

I would ask that nobody closes this issue, only myself when I confirm that a solution has worked.

https://github.com/arendst/Sonoff-Tasmota/issues/3857#issuecomment-423601308

L00000813, 2018-10-06:
I now also have a 1 Gang T1 UK switch and it does exactly the same thing?

andrethomas2, 2018-10-06:
Please follow the path given to you when https://github.com/arendst/Sonoff-Tasmota/issues/3857#issuecomment-423601308 was closed.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-06 by andrethomas2