arendst/Tasmota #3989: Development: switchmode 3/4 not working

markusn1, 2018-10-06:
I am using the Shelly 1 and uploaded the latest sonoff.bin from thehackbox (compiled 20181006 11:02 GMT+2). I found out that switchmode 1/2 work fine and trigger the relay directly, 3/4 do not seem to do anything. 

status 0:
`12:10:45 MQT: stat/shelly1/STATUS = {"Status":{"Module":46,"FriendlyName":["Sonoff"],"Topic":"shelly1","ButtonTopic":"","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:10:45 MQT: stat/shelly1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://tasmota.chanoa.de/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:39","StartupUTC":"2018-10-06T10:56:06","Sleep":0,"BootCount":24,"SaveCount":139,"SaveAddress":"FA000"}}
12:10:45 MQT: stat/shelly1/STATUS2 = {"StatusFWR":{"Version":"6.2.1.11","BuildDateTime":"2018-10-06T11:00:47","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:10:45 MQT: stat/shelly1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["private@chanoa.de",""],"TelePeriod":300,"SetOption":["0000880B","55818000","00000000"]}}
12:10:45 MQT: stat/shelly1/STATUS4 = {"StatusMEM":{"ProgramSize":479,"Free":524,"Heap":15,"ProgramFlashSize":1024,"FlashSize":2048,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","000003C0"]}}
12:10:45 MQT: stat/shelly1/STATUS5 = {"StatusNET":{"Hostname":"shelly1","IPAddress":"192.168.40.15","Gateway":"192.168.40.62","Subnetmask":"255.255.255.192","DNSServer":"192.168.40.152","Mac":"BC:DD:C2:8B:18:E4","Webserver":2,"WifiConfig":5}}
12:10:45 MQT: stat/shelly1/STATUS6 = {"StatusMQT":{"MqttHost":"fhem.chanoa.de","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8B18E4","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:10:45 MQT: stat/shelly1/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 06 11:10:45 2018","Local":"Sat Oct 06 12:10:45 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:57","Sunset":"18:19"}}
12:10:45 MQT: stat/shelly1/STATUS10 = {"StatusSNS":{"Time":"2018-10-06T12:10:45","Switch1":"ON"}}
12:10:45 MQT: stat/shelly1/STATUS11 = {"StatusSTS":{"Time":"2018-10-06T12:10:45","Uptime":"0T00:14:39","Vcc":3.228,"POWER":"ON","Wifi":{"AP":1,"SSId":"private@chanoa.de","BSSId":"4E:5E:0C:B9:03:40","Channel":1,"RSSI":100}}}`

ascillato2, 2018-10-06:
Hi,

I could not reproduce your issue. It is working fine for me.

Could you please, show your GPIO configuration, also the outputs of your console (using weblog 4 in order to have more debug messages) and how you have your switches connected to your device.

Thanks

markusn1, 2018-10-06:
I dont use any GPIO ports. I connected a momentary switch do the SW terminal on the Shelly 1. The console does not show anything if i configure switchmode 3 or 4 and press the button. Like i said, switchmode 1/2 work just fine

arendst, 2018-10-06:
It might be caused by switch pull-up issue.

If you compile yourself could revert a change made for the shelly2 by making the following change in sonoff.ino line 2310

From:
```
void GpioSwitchPinMode(uint8_t index)
{
  if (pin[GPIO_SWT1 +index] < 99) {
//    pinMode(pin[GPIO_SWT1 +index], (16 == pin[GPIO_SWT1 +index]) ? INPUT_PULLDOWN_16 : bitRead(switch_no_pullup, index) ? INPUT : INPUT_PULLUP);

    uint8_t no_pullup = 0;
    if (bitRead(switch_no_pullup, index)) {
      no_pullup = (Settings.switchmode[index] < PUSHBUTTON);
    }
    pinMode(pin[GPIO_SWT1 +index], (16 == pin[GPIO_SWT1 +index]) ? INPUT_PULLDOWN_16 : (no_pullup) ? INPUT : INPUT_PULLUP);
  }
}
```
To:
```
void GpioSwitchPinMode(uint8_t index)
{
  if (pin[GPIO_SWT1 +index] < 99) {
    pinMode(pin[GPIO_SWT1 +index], (16 == pin[GPIO_SWT1 +index]) ? INPUT_PULLDOWN_16 : bitRead(switch_no_pullup, index) ? INPUT : INPUT_PULLUP);
/*
    uint8_t no_pullup = 0;
    if (bitRead(switch_no_pullup, index)) {
      no_pullup = (Settings.switchmode[index] < PUSHBUTTON);
    }
    pinMode(pin[GPIO_SWT1 +index], (16 == pin[GPIO_SWT1 +index]) ? INPUT_PULLDOWN_16 : (no_pullup) ? INPUT : INPUT_PULLUP);
*/
  }
}
```

markusn1, 2018-10-06:
It seems like that did the trick! Switchmode 3 & 4 now work as expected!

arendst, 2018-10-06:
Ok. I will update it tomorrow.

arendst, 2018-10-07:
Could you be so kind to test if the fix solves your issue and report back please.

markusn1, 2018-10-07:
Just compiled the latest development, flashed it and it is still working fine! Thank you Theo!

closed 2018-10-07 by arendst

TheEggi, 2018-10-11:
Does the latest DEV version now work with the Shelly 1 AND 2?

arendst, 2018-10-11:
Yes