arendst/Tasmota #3990: Deep Sleep / Low power consumption

nosaj66au, 2018-10-06:
Hi Guys,
     The house is sitting with about 75 tasmota devices and now I have moved to monitoring water take level using the waterproof version of  JSN-SR04T. I'm looking at remotely locating it from the house about 40m and using solar power and a battery. I would like a means of report by exception or time interval with deep sleep between reports. I have used the wiki info on battery power devices, however the best I can get is just over 1 day with a 2000mA (18650) battery on a wemos 18650 board and I haven't yet connected the ultrasonic. I was hoping for around 3 days but I believe the radio being left on is killing me. Current setting are Telemetry time 20 minutes and the sleep time at 250ms.
Maybe this can be achieved in a different manor?

Jason

nosaj66au, 2018-10-06:
I think I have just found the problem is Sleep still disable in newer build ?

ascillato2, 2018-10-06:
Hi,

Deep sleep is a very nice feature but it is not supported by Tasmota due to be outside the actual scope of the project (home automation - see readme), so, Tasmota needs to be up and running in order to be responsive to switches, etc.

Nevertheless, Tasmota do support light modem sleep that will save a lot of energy (see wiki). But that feature only works using esp8266 core library version 2.3.0 by now. You need to have a firmware compiled with that. If you use any other core, Tasmota has sleep feature disabled because it makes the SDK to hung the device.

So, if you compile by yourself, use core 2.3.0 to have sleep feature. Or just use any of the precompiled bins that are compiled with 2.3.0.

Thanks.

If you have further questions, please find us in the Tasmota Discord chat.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-06 by ascillato2

Asuanin, 2019-05-28:
Hi,
what excludes battery powered devices from home automation?
For example, battery powered thermometers in any room or soil moisture meter are part of home automation. If I need to measure temperature only at a certain interval (5 or 10 minutes), not continuously, then deep sleep for the battery is best.

magiva, 2019-08-17:
i totally agree,  part of hone automation is data collection to automate other things.  deep sleep is very much needed

NightDragon1, 2019-09-25:
I absolutely support and agree to a deep sleep feature included in Tasmota. Especially for hoem automatisation together with todays IOT deep sleep is important!!

meingraham, 2019-09-25:
Give the stefanbode Tasmota fork a try.