arendst/Tasmota #3991: adding CSL (aplic WDP 303075) Socket

Phelagor, 2018-10-06:
Hello, I received a CSL wifi smart plug https://www.amazon.de/dp/B07CNWVNJ2, which is based on a ESP8266. 
It is pretty similar to the sonoff devices.
I already made some pictures of the pins and successfully flashed tasmota on that socket. Via the 'generic' Module I could already check the relais, button and LED functions. 
I also added the module locally to tasmota and flashed the firmware update, and almost everything works as expected, but not the build in HLW8012, which is used to measure current, voltage and power.

The device reads some values, but they are totally implausible. I tried to calibrate the device as described in https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow-and-Pow-R2 but the values are still not correct.

I add some pictures for the pins of the ESP8266 Module, and can describe the proccess of flashing the device also might create a PR, but the software part isn't done yet, at least if the power monitor should also work (everything else works as the LED, the Button and toggeling the Relais).

Since I am totally new to tasmota and don't know the full source code, I don't know where to find stuff about the HLW8012, which if I am informed correctly is also used in the sonoff POW 2. I am reading some information about the HLW8012 rn, and might measure values and stuff as far as possible, but might need some hints where to look for calculation part in the software.

I also add a screenshot of 'generic' module configuration (which doesnt support HLW pins), for those who just want to use the relais, button and LED.

PCB-Pins
[image]
[image]
[image]

'generic'-Module config
[image]

andrethomas, 2018-10-06:
Hi,

The HLW8012 is used in the Sonoff POW, not the POW R2.

So it might be possible to create a new template based on the original POW as it uses the HLW8012 which is PWM based. The POW R2 uses a different chip.`

ascillato, 2018-10-06:
@Phelagor 

Please,

Try first to adapt the sonoff pow template (sonoff_template.h file) to your device to test the measurements.

Phelagor, 2018-10-06:
Ok, thank I'm gonna try to adapt the sonoff pow to the device. Thanks for the hints.

Phelagor, 2018-10-06:
> @Phelagor
> 
> Please,
> 
> Try first to adapt the sonoff pow template (sonoff_template.h file) to your device to test the measurements.

So were you talking about just to add the template? I already did that locally. But the values I get from the HLW are just not plausible...
`  { "AplicWDP303075",   // Aplic WDP 303075 (ESP8285 - HLW8012 Energy Monitoring)
                       // https://www.amazon.de/dp/B07CNWVNJ2
     0,                 // GPIO00 
     0,                 // GPIO01 
     0,                 // GPIO02 
     GPIO_KEY1,         // GPIO03 Button
     GPIO_HLW_CF,       // GPIO04 HLW8012 CF (power)
     GPIO_HLW_CF1,      // GPIO05 HLW8012 CF1 (current/voltage)
     0, 0, 0, 0, 0, 0,  // ?
     GPIO_HLW_SEL,      // GPIO12 HLW8012 CF Sel output (CF)
     GPIO_LED1_INV,     // GPIO13 LED
     GPIO_REL1,         // GPIO14 Relay SRU 5VDC SDA (0= Off, 1 = On )
     0,                 // GPIO15
     0, 0
  }`

ascillato2, 2018-10-06:
For testing proposes, will be better just to change the GPIO definition of the sonoff pow template to match your device.

Phelagor, 2018-10-06:
> For testing proposes, will be better just to change the GPIO definition of the sonoff pow template to match your device.

I just changed the GPIO definitions of the sonoff pow template to those I posted above. But still the values are just not right. 
Also when trying to calibrate via console commands I get those values:

> 18:21:12 CMD: PowerSet 51
18:21:12 RSL: Received Topic /PowerSet, Data Size 2, Data 51
18:21:12 RSL: Group 0, Index 1, Command POWERSET, Data 51
18:21:12 RSL: RESULT = {"PowerCal":4598}
18:21:13 CFG: Saved to flash at F7, Count 171, Bytes 3584
18:21:20 CMD: VoltageSet 228
18:21:20 RSL: Received Topic /VoltageSet, Data Size 3, Data 228
18:21:20 RSL: Group 0, Index 1, Command VOLTAGESET, Data 228
18:21:20 RSL: RESULT = {"VoltageCal":32023}
18:21:21 CFG: Saved to flash at F6, Count 172, Bytes 3584
18:21:26 WIF: Checking connection...
18:21:26 WIF: Connected
18:21:28 CMD: CurrentSet 220
18:21:28 RSL: Received Topic /CurrentSet, Data Size 3, Data 220
18:21:28 RSL: Group 0, Index 1, Command CURRENTSET, Data 220
18:21:28 RSL: RESULT = {"CurrentCal":76}

The values under load then look ok, but when removing the load it shows wrong voltage and voltage values fluctuate really bad.
under load
[image]
no load
[image]

ascillato, 2018-10-06:
try in the console `setoption21 1`

Phelagor, 2018-10-06:
OH. I think I found the issue.
I always checked the Values on the webinterface, and it shows only Voltage when no load connected or the relais is open.
BUT when checking the whole Sensorlogs in console I see it reports a current..
And the current is the value I calibrated the device for.

It seems like Voltage and current are just swapped! I dont know exactly what tasmota is calculating or how the values are proccessed, but it seems like it just shows some weird voltage when nothing is connected / relais is open, because of the wrong signals it gets from the HLW when expecting the volatage signals...

I just have to figure out **how do I swap the values of GPIO_HLW_SEL?** so it gets the right values when expecting voltage / current.

Phelagor, 2018-10-06:
To test it, I just changed HLW_SEL_VOLTAGE 1 to HLW_SEL_VOLTAGE 0 in xnrg_01_hlw8012.ino, and checked that firmware.

It now shows the correct values. I might need some help to change the code properly to fit the conventions.
Another Version I didnt build yet, I just added 

> // HLW8012 based (APLIC_WDP303075)
#define A_HLW_SEL_VOLTAGE         0
....
else if (APLIC_WDP303075 == Settings.module){
    hlw_power_ratio = HLW_PREF;
    hlw_voltage_ratio = HLW_UREF;
    hlw_current_ratio = HLW_IREF;
    hlw_ui_flag = A_HLW_SEL_VOLTAGE;

Might check if that works, but also might need someone to check where I might have to change other stuff in case those values are used somewhere else...

andrethomas, 2018-10-06:
What happens if you just swap the pins on the template?

Phelagor, 2018-10-06:
Nah, the pins on the template are for the two datapins of the HLW, so either for Power measurement (CF) or for voltage/current (CF1). So swapping CF and CF1 would just result in power data going to be interpreted as voltage.
There is a select-mode pin on the HLW to change wich values you get on CF1, either voltage OR current. But if the curcuit arround the HLW is different than the one on sonoff Pow (maybe permanently pulled high instead of low), you need different signal to change the data-stream on CF1.

arendst, 2018-10-06:
With the info here I will make changes and release tomorrow. Thx flr testing.

Phelagor, 2018-10-06:
Oh, I was just trying to create a pull request, but I still fight with git and the pull request feature 😕

Jason2866, 2018-10-06:
Do you have photos from the Smart plug where the PCB and the AC part can be seen?
I am interested how the build quality is. What is your opinion? Well done?
Thank you!

Phelagor, 2018-10-06:
No, I dont have any pics of that part atm.
I cant say much about the quality, just the soldering quality was pretty bad, especially at the main wires.
I just checked one of the bigger capacitor near the mains, and it said 640V iirc.
But compared to the Gosund Plugs, which I flashed just yesterday, it looked almost the same.

Phelagor, 2018-10-06:
Added a wiki page
https://github.com/arendst/Sonoff-Tasmota/wiki/CSL-(aplic-WDP-303075)

ascillato2, 2018-10-06:
Closing issue as PR #3996 was merged and now it is part of Tasmota. Thanks a lot for all the contributions.

arendst, 2018-10-07:
The fix is a bit refurbished to accommodate future changes.

Could you be so kind and test if it still does what it should do and report back please.

Phelagor, 2018-10-07:
I tested it and it works as expected. No problems with that build.

closed 2018-10-07 by arendst