arendst/Tasmota #3997: Servo control in Tasmota

pwyrzyk, 2018-10-06:
Is it possible to add servo controlled by MQTT to tasmota?
I did some code which works as stand alone but I would be great to integrate with Tasmota build.
Thanks.

Code (servo part - w/o mqtt and Wifi code)

#include <Servo.h>
int servoAngle;

Servo myservo;
WiFiClient espClient;
PubSubClient client(espClient);

void callback(char* topic, byte* payload, unsigned int length) 
{
  String value = "";
  for(int i=0;i<length;i++)
  {
    if((char)payload[i]>194||(char)payload[i]<0)
    break;
    value += ((char)payload[i]);
    Serial.print(value);//print the rotation angle
    Serial.println();
  }
  servoAngle = value.toInt();
  Serial.println(servoAngle);
  myservo.write(servoAngle);
}//end callback

pwyrzyk, 2018-10-06:
Yes, but no success with PWM. But I'm not familiar with PWM on Tasmota. Is there an option to control servo?

andrethomas, 2018-10-06:
For servo's the PWM frequency needs to preferably be 50hz - I am unsure if the pwm frequency can be defined in tasmota for a normal GPIO but you could add this over I2C https://github.com/arendst/Sonoff-Tasmota/wiki/PCA9685

ascillato, 2018-10-06:
In the wiki at commands https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management
using the Tasmota console, you can define the PWM frequency. Can you try if the minimal (100Hz) is working for you?

For that:

* Set the GPIO you want to use with PWM as PWM1
* Go to console and type:
```
SetOption15 1
PwmFrequency 100
PWM1 10
```
and then change the PWM1 from 0 to 1023

pwyrzyk, 2018-10-06:
Yes, I was stacked with servo+tasmota on PWM low frequency - there is the limitation =>100Hz. Maybe, that it could be changed in the code (there is a line with freq range):

 _if ((1 == payload) || ((payload >= 100) && (payload <= 4000)))_ 

 but I'm afraid that it could be also some HW limitation.

 I will check I2C option. Regardless, it think that using **servo.h** library and the code I copied in my original post Tasmota could control servo via MQTT in very easy way.

pwyrzyk, 2018-10-07:
OK Solved.
I've changed the minimum frequency from 100 to 50Hz in the source code:
else if (CMND_PWMFREQUENCY == command_code) {
if ((1 == payload) || ((payload >= **50**) && (payload <= 4000)))

then from console used commands;
SetOption15 1
PwmFrequency 50

PWM1 - command doesn't work for me but DIMMER does.
DIMMER supports resolution from 1 to 12 for 180 degrees of my servo rotation (SG90), but it is enough for me.

Thanks for advice.

Jason2866, 2018-10-07:
If you want to use PWM you have to set SetOption15 0

Jason2866, 2018-10-07:
Please close this issue. Your question seems to be answered.

closed 2018-10-07 by ascillato2

ezraholm50, 2019-01-27:
@pwyrzyk Could you explain your servo setup?

I have a NodeMCU board with an ugly .ino file for mqtt
What i do now:
ON = servo.write 180; delay xxxx
OFF = servo.write 0; delay xxxx

This is on a continuous servo that opens my blinds with a rope attached to 2 pully's

I'd like to do this with Tasmota, way more reliable... Any thoughts?

pwyrzyk, 2019-01-31:
I would try to do this in this way in Tasmota:
1. set up configure module/module typ = 18 Generic
2. define nodeMcu PWM pin f.ex D1(GPIO5) as PWM1
3. from console use: SetOption15 1
4. from console or via mqtt use: dimmer 0-100  - to control servo in available angle

You can experience the PWM frequency issue. By default it 880hz and you can change it from 100 up to 4000hz (pwmfrequency xx command). My servo works with 50hz. In new Tasmota releases in sonof.h there is PWM_MIN (default 100). I changed it to 50 and compiled and then I can change the pwm frequency from tasmota console to 50hz using pwmfrequency 50.
I hope it will work for you.

andrethomas, 2019-01-31:
[image]

50hz is spot on for a 20ms duty cycle.

As you can see above a normal servo is going to do -90 deg on ~1ms (i.e. PWM value of 5) and +90 deg on ~2ms (i.e. PWM value of 10)

andrethomas, 2019-01-31:
@gitolicious Just make sure the servo's support high frequency, otherwise you risk failure at some point. Some of the newer digital servo's do not mind this.

gitolicious, 2019-02-01:
To be honest, it was trial&error without the slightest clue on how all this works, and at 1€/pc I just gave it a go. I will try 50 Hz as well and report back, but according to your drawing above, only dimmer values between 5 and 10 will be usable?

andrethomas, 2019-02-01:
That is correct in terms of standard servo requirements. It needs a pulse width of 20ms (50hz) and a duty cycle of 1 millisecond to 2 millisecond as described above to determine the position of the servo. The servo uses this pulse length to determine what needs to be done to achieve the same pulse width on the internal electronics of the servo and it is with this simple comparative measure that it achieves the positioning.

So in short servo's were not originally intended to work with PWM but rather PPM which is defined as in the picture above.

gitolicious, 2019-02-01:
Thanks for the info. Will check with my logic analyser.
Maybe then an optional dimmer value translation in tasmota where `Dimmer 0` is translated to 5, `Dimmer 100` to 10, and all values in between accordingly would be a nice addition.

ezraholm50, 2019-02-01:
Thanks! but will this work for a continous servo? Can i set the time anywhere?

ezraholm50, 2019-02-01:
Ok now this works great (100 hz), i can control it via MQTT with HASS automations so i'd just a 
mqtt: dimmer 5
delay: xx sec
mqtt: dimmer 0

10 is one way / 20 the other

This is a great workaround for my continous horizontal curtains 50/50, just need to know the close/open time.

Although, i've tried compiling it myself for over 24 times... everytime it fails... Using arduino IDE could you provide me the sonoff.bin you've compiled?

Or point me in which file i need to edit:
_ else if (CMND_PWMFREQUENCY == command_code) {
if ((1 == payload) || ((payload >= 100) && (payload <= 4000)))_

Since i found something that looked like it but it was gray in Arduino IDE, not sure if it was the right section.

Thanks guys

gitolicious, 2019-02-02:
It should also be possible to use tasmota's `Backlog` command for your delay, right? 

And your compilation issues should probably be discussed elsewhere (separate topic).

ezraholm50, 2019-02-02:
Thats awesome backlog with delay set thx