arendst/Tasmota #3999: SerialMonitor from Sonoff pow is: NULNUL

shayshtern, 2018-10-06:
I did reflash with atom and its was all ok flashing.
after that the device isn't connecting to my Wi-Fi and when I try to get information with Atom's Serial Monitor I got: 'NULNULNULNULNULNULNUL'
Photo: 
[image]
What Can I do? 
Is there "hard flash" or something similar?
Thanks !

Jason2866, 2018-10-07:
Disconnect it. Reasemble, connect it to mains. It will probably work
POW uses serial for measurement. You cant use serial...
See https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow-and-Pow-R2

ascillato, 2018-10-07:
Serial log is used by the power measurement chip. Can not be used for monitoring.

Please, close this issue. Thanks

shayshtern, 2018-10-07:
At the end I flash it with this tool: https://github.com/marcelstoer/nodemcu-pyflasher/releases
Than open a serial monitor from Atom and set the WiFi and password with sending commands! 
Thank anyway!

closed 2018-10-07 by ascillato2