arendst/Tasmota #4000: WEB Commands for SONOFF RF Bridge 433 WiFi

boklubre, 2018-10-07:
Hello,

I try to control my SONOFF RF Bridge 433 (Firmware Tasmota 6.2.1) also via Webcommands.
At the moment I control them by my MQTT-Server and FHEM for Intertechno Plugs and it´s works very fine.
But I also need a access via Webcommands. I have tried the example on the WIKI. But it doesent work.
In FHEM I can see that the ON Command for the Plag is #51015 and the OFF Command is #51014.

I have tried the following: 

http://192.168.233.60/cm?&user=xxxxpassword=xxxx&cmnd=RfCode%20#51015

It doesent work. Any Idea?

Cheers

Boris

Jason2866, 2018-10-07:
You have to send the timings too.
For example
http://192.168.233.60/cm?&user=xxxxpassword=xxxx&cmnd=backlog RfSync 4860; RfLow 140; RfHigh 470; RfCode #51015

Jason2866, 2018-10-07:
Why via web and not via mqtt?

boklubre, 2018-10-07:
I want to use in the future without a special MQTT Server.
I only want to have one device to maintaince (CCU3) and for the IT Plugs I dont need a bidirectinal connection.

boklubre, 2018-10-07:
I have found out that it works with decimal code (without sending the timmings):

http://192.168.233.60/cm?&user=xxxx&password=xxxx&cmnd=RfCode%20331796

closed 2018-10-07 by boklubre