arendst/Tasmota #4001: RF Bridge stops sending/receiving after a few hours

simo878, 2018-10-07:
**Describe the bug**
RF Bridge stops sending and receiving RF signals after a few hours, until a hard reset is performed.

After a successful transmission, the console usually displays:

14:15:30 MQT: stat/RF_Bridge/RESULT = {"RfRaw":"ON"}
14:15:30 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}

but when the RF Bridge is in this failed state it no longer displays the second line, it still acknowledges the commands to send with the RfRaw:ON after each attempt to transmit 

I have upgraded the RF chips firmware as described in the tasmota wiki (Portisch firmware)

example strings that i am transmitting:

AAB017041401DB01170226226A1011001010101010101010101355
AAB0250514024D02B6011901D61FE02000202020202020000000000000002020202020200000000455

i only use the bridge to transmit, but also tested it ability to receive when it is in the failed mode, and it also does not receive signals.

soft resetting via the console does not fix the problem (reset 1), only a hard reset, by pulling the power will make the bridge work again.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff RF Bridge (R2 same model pictured on the website - https://www.itead.cc/sonoff-rf-bridge-433.html)
- [x] _Development/Compiler/Upload tools used_ :  ESPEasy flasher - Tasmota 6.2.1 - sonoff.bin
- [x] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
14:03:48 CMD: status 0
14:03:48 MQT: stat/RF_Bridge/STATUS = {"Status":{"Module":25,"FriendlyName":["RF_Bridge"],"Topic":"RF_Bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
14:03:48 MQT: stat/RF_Bridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"1T11:58:29","StartupUTC":"2018-10-06T01:05:19","Sleep":0,"BootCount":41,"SaveCount":59,"SaveAddress":"F5000"}}
14:03:48 MQT: stat/RF_Bridge/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:03:48 MQT: stat/RF_Bridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Archer 2.4",""],"TelePeriod":300,"SetOption":["00008029","55818000","00000000"]}}
14:03:48 MQT: stat/RF_Bridge/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
14:03:48 MQT: stat/RF_Bridge/STATUS5 = {"StatusNET":{"Hostname":"RF_Bridge","IPAddress":"10.1.1.111","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"84:0D:8E:42:C2:9C","Webserver":2,"WifiConfig":5}}
14:03:48 MQT: stat/RF_Bridge/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.20","MqttPort":1883,"MqttClientMask":"RF_Bridge","MqttClient":"RF_Bridge","MqttUser":"mqttpi","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:03:48 MQT: stat/RF_Bridge/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 07 13:03:48 2018","Local":"Sun Oct 07 14:03:48 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:58","Sunset":"18:17"}}
14:03:48 MQT: stat/RF_Bridge/STATUS10 = {"StatusSNS":{"Time":"2018-10-07T14:03:48"}}
14:03:48 MQT: stat/RF_Bridge/STATUS11 = {"StatusSTS":{"Time":"2018-10-07T14:03:48","Uptime":"1T11:58:29","Vcc":3.141,"Wifi":{"AP":1,"SSId":"Archer 2.4","RSSI":100,"APMac":"FC:EC:DA:11:BC:2A"}}}

Jason2866, 2018-10-07:
Your problems seems related to this
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/32
Feel free to open an issue on Portisch site.
Since the issue is NOT releated to Tasmota firmware please close this issue.
Thank you

closed 2018-10-07 by ascillato2

simo878, 2018-10-15:
After discussing and troubleshooting with Portisch, they seem to think the bug is with tasmota. see

https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/32

Jason2866, 2018-10-15:
Confirmed. Fix is on the way...

andrethomas2, 2018-10-15:
Done, thanks for reporting.