arendst/Tasmota #4003: CE Smart Home Wifi Smart Dimmer Light Switch (TYWE1S) Model WF500D

badabing2005, 2018-10-07:
Has anyone had success with:
CE Smart Home Wifi Smart Dimmer Light Switch ([TYWE1S](https://fccid.io/2ANDL-TYWE1S/User-Manual/User-Manual-3695972)) Model [WF500D](https://fccid.io/ZZH-WF500D/User-Manual/Users-Manual-3846958#download)

From the same series of this [Wifi](https://github.com/arendst/Sonoff-Tasmota/wiki/CE-Smart-Home---LA-WF3-Wifi-Plug-(TYWE2S)) Plug which is already supported.

Purchased at Costco, a [pack of two](https://www.costco.ca/Wi-Fi-Smart-Dimmer-Light-Switch%2c-2-pack.product.100417574.html) for $39.99 CDN

The manual can be found [here](https://fccid.io/ZZH-WF500D/User-Manual/Users-Manual-3846958#download)

# Product
[image]

Here are some disassembly pics and notes.
[image]

[image]

They've made it particularly difficult to take it apart.
You need to de-solder or clip the connections.

[image]

[image]

[image]

[image]

[image]

Looks like the same PCB will be used for the 3-way versions.
[image]

Thanks
BB

ascillato, 2018-10-07:
Hi,

Dimmers are not supported yet.

Please see #469 for further information.

Thanks.

andrethomas2, 2018-10-07:
Closing this issue as it has been answered. Thank you for your valued inputs.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-07 by andrethomas2

qingz2004, 2018-10-25:
It can probably be supported by switching UART port.

qingz2004, 2018-10-28:
It is supported.
Please see [#469](https://github.com/arendst/Sonoff-Tasmota/issues/469) for further information.
Here is the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer).
Flash it the same way as the other ESP8266 product.
[image]

badabing2005, 2018-10-28:
Thank you @qingz2004 
Just to be sure, the relay portion of the device does not need to be connected to flash
The black on the right is connected to U0RX
The black on the left if connected to GPIO 0
The red above it is connected to U0Tx
and the red at the bottom is the 3.3v (obviously)

Can I flash the latest [sonoff.bin](https://github.com/arendst/Sonoff-Tasmota/releases) 
Probably not as it is dated Sep 5th
Which build should I flash?
Can I set all the settings afterwards through serial or do I need special defines / settings in the source and build from that?

Thanks for the awesome work.
BB

qingz2004, 2018-10-28:
@badabing2005 The connection is right, plus I used the GND pad from bottom right corner for easy soldering.
You can not use the sonoff.bin from Sep 5th. The code (Tuya Software Serial) is just added in the development branch yesterday by @chaosmaster.
You may have to compile it from the development branch.

badabing2005, 2018-10-28:
Thanks @qingz2004 

Ok Got the Development branch, built sonoff using VS Code (Platform IO IDE)
flashed firmware.bin using ESPEasy Flasher
Then in Termite
All I get is:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff38f8
~ld
```

And I also don't see Direct Wifi for Sonoff.

I'm sure I'm doing something wrong, if you could kindly point out where, it would be nice.
Thanks

badabing2005, 2018-10-28:
Even flashing [this](https://github.com/arendst/Sonoff-Tasmota/files/2521367/tuya-software-serial.zip), I get the same.

qingz2004, 2018-10-28:
I set my own WiFi in the user config.

badabing2005, 2018-10-28:
I must be doing something wrong.
Entered my wifi settings in sonoff\my_user_config.h
Rebuild
Flashed
[image]

Still the same thing, It's not connecting to wifi and serial connection shows
constantly scrolling
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff3948
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff3948
~ld
```

:(

qingz2004, 2018-10-28:
I got the same with one of my dimmer.
Try with another one.

badabing2005, 2018-10-28:
So this one is considered paperweight?

I'll try with the other, but would hate to lose both.
Taking these apart is such a Pita, did you manage to desolder and detach the jumper pins?
For the Voltage regulator, it wasn't bad, but the 2mm jumper pins, I had to clip them (see pic above), now I'm waiting for the replacement jumpers to arrive.

Thanks for all your assistance.
BB

badabing2005, 2018-10-28:
Not sure if [this](https://forum.arduino.cc/index.php?topic=408876.msg3317127#msg3317127) would apply to us.

qingz2004, 2018-10-28:
No, I didn't desolder the jumper pins. Just desolder the Triac and bend the jumper.

badabing2005, 2018-10-28:
Thanks, I'll do that on the second unit :)

qingz2004, 2018-10-28:
I flashed three, two of them are good. I will debug the bad one later when I have time.

badabing2005, 2018-10-28:
> 
> 
> > So this one is considered paperweight?
> 
> You can still use it without WiFi. I can not connect to it with the stock firmware. So to me, it's not worse.
> Can you connect to it with the CE app?

Thanks,
I'll bend the jumper this time
I won't have a chance today, but I'll give it a shot tomorrow.

Thanks for your help, let me know if you manage to find out what's wrong with your other unit.
BB

badabing2005, 2018-10-29:
Just got back and couldn't wait,
I flashed the second Dimmer.
Strangely I got absolutely nothing on the serial after flashing.
I couldn't even get status.
Surprisingly, it connected to my wifi, and I was able to add more settings through webui
I chose Tuya Dimmer (54)
Do I need to set anything else?

I can get the console in webui, but serial connection I get nothing, are these at 115200 baud?
Any idea why I get nothing on serial? if there was anything wrong with my serial connection, I wouldn't have been able to flash.

What's next, just close the box and put in in wall?
I suppose update flashes can now all be done OTA, and I can desolder the serial pins?
How do I test this before connecting to mains?
I have green LED constantly flashing, but I see in the dimmer thread that you have that issue as well and working with Joel.

Thanks for all the hard work.
BB

qingz2004, 2018-10-29:
Check the wiki for settings.

badabing2005, 2018-10-29:
Thanks,
Done and settings updated.
Was that supposed to address the serial issue? if it was, it hasn't.

Good night.
BB

badabing2005, 2018-10-29:
Ah ok, it's not only mine, you have the same issue.

Thanks
BB

qingz2004, 2018-10-29:
I think you should get the messages if you connect to GPIO13 and 15 (U1Rx/TX).

e2m32, 2019-06-22:
I have a Moes DS01 using this same chip (TYWE1S). It flashed successfully with the pre-compiled bin using ESPEasy (FlashESP8266), but I couldn't termite in to put in my SSID info. I tried connecting to U0TX/RX and U1TX/RX with no luck. It also didn't go into WIFICONFIG mode where it broadcasts an SSID to connect to. As a last hurrah, I tried putting it back together (connecting the ribbon cables) and connecting it to mains power. THAT WORKED! I was able then to connect to the device over WIFI to configure it. Looks like its all working now :-)

As you can see, I have a mess of wires going on here ;-)
[image]

arduino43, 2019-06-23:
I bought one of the Moes Dimmers as well, but can not flash it.

I've connected UX0,TX0 and GPO0 to ground, but for some reason it isn't allowing flash. Is there any particular changes you made to this setup that are not posted here?

arduino43, 2019-06-23:
Nevermind - I had UX0 connected to the wrong pin.

arduino43, 2019-06-23:
> I have a Moes DS01 using this same chip (TYWE1S). It flashed successfully with the pre-compiled bin using ESPEasy (FlashESP8266), but I couldn't termite in to put in my SSID info. I tried connecting [...]

I've got it running, but it doesnt respond using the Tuya Dimmer (54) template. Did you have any additional configuration? I've also set SetOption34 2 but it still does not work.

jmw6773, 2019-08-16:
arduino43,
  I have the same switch and set it up today. I don't know if you got it working, but try using the template found [in the wiki.](https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer)

  You'll also need to reset SetOption34 to 3 again so the ESP can communicate with the MCU on the back.

arduino43, 2019-08-17:
That worked, the flashing red LED was frustrating.

The LED is now solid white - Thanks!

jmw6773, 2019-08-19:
arduino43,
  I found that flashing digiblur's fork of Tasmota worked much better for this dimmer. Before when I sent a value to dim to, it would set that value and then change it's value to a lower level. Digiblur's fork fixed this.

  [https://github.com/digiblur/Sonoff-Tasmota](https://github.com/digiblur/Sonoff-Tasmota)

wyxsmarthome, 2020-10-20:
I flash mine with tuya-convert. Set the module like below, it works. I am still having problem to control it with "tasmota-connect" though.
[image]

papapep514, 2020-11-02:
@wyxsmarthome remember to enter command "SetOption19 1"
@jmw6773 , were you talking about the Model WF500D using the Digiblur's fork or the Moes Dimmer using Digiblur fork?
I tried using the template and infos on "blakadder ce_smart_home_CFW500D-3W", but neither are working ...

jmw6773, 2020-11-03:
@papapep514, I was recommending the Digiblur fork for the Moes Dimmer. However, I believe that Digiblur's changes were merged into the main Tasmota branch a few releases ago. I'm using the normal Tasmota release on my dimmer without issue now.

I don't have a WF500D to test with, sorry.

marcoboutch, 2021-07-12:
command for me is 

TuyaMCU 21,3
DimmerRange 10,255
TuyaMCU 11,1

https://www.youtube.com/watch?v=_3WW4NVYHrU

aaronmtg, 2021-12-30:
Somehow I missed the step about clipping the header at the top, Why do we need to do that?

I have tasmota up and running but can not control the connected lights. (turning on/off does nothing)