arendst/Tasmota #4005: Where to set the HOSTNAME?

aaamitsingh, 2018-10-08:
Remembering mac_id of very tough for every module so I want to set different Hostname for the different sonoff module so that I can easily identify the which module is up and down from the IP scan tools. But don't know where to change. Please help..

andrethomas, 2018-10-08:
From web ui, Configuration|Configure Wifi

[image]

closed 2018-10-08 by andrethomas2

ramanraja, 2023-11-06:
I tried to remove the %4d at the end of the host name since I don't want it. Also tried to give the hostname %s command from the console. But every time the 4 digit appendage was reappearing. How to get rid of that ?

sfromis, 2023-11-06:
Those usages of `%` are for the default naming scheme, not for you to adapt on. Preferable to set your own full host name without getting confusion around how those `%` tags are not straightforward.

(In general, questions should be asked in the "Discussions" tab, not in "Issues")

ramanraja, 2023-11-08:
Thank you @sfromis.
For the sake of closure and to help late comers: I discovered that the same issue has been discussed from 2017 to 2022 in different threads. The most comprehensive answer is found in https://github.com/arendst/Tasmota/issues/142