arendst/Tasmota #4007: Serial Number for emulated "Wemo"-Device

TMHEINE, 2018-10-08:
Hello,
I use this firmware on 2 Blitzwolf SHP2 plugs. And I want to include them into my Solarlog via the WeMo emulation mode.
But during configuration, the Solarlog asks me for the last 3 digits of the serial number on the "WeMo" plug.
How can I get these numbers? Or can I also emulate a serial number for this device?
Is there an solution, or is it just impossible to include them?

Kind Regards
Tom

Jason2866, 2018-10-08:
With a quick look in xplg_wemohue.ino it should from the ESP ChipId

```
String WemoSerialnumber()
{
  char serial[16];

  snprintf_P(serial, sizeof(serial), PSTR("201612K%08X"), ESP.getChipId());
  return String(serial);
}

String WemoUuid()
{
  char uuid[27];

  snprintf_P(uuid, sizeof(uuid), PSTR("Socket-1_0-%s"), WemoSerialnumber().c_str());
  return String(uuid);
}

```

closed 2018-10-08 by ascillato2

TMHEINE, 2018-10-08:
Hi,
How can I get the "WemoSerial number" via console.
The last 3 Digits of the ESP Chip-ID(Information-site) are not working in my case.
Or ist an original Wemo device providing the Serial number in a different way?