arendst/Tasmota #4008: Not disabled the option PowerOnState4

skop1986, 2018-10-08:
It is not possible to disable PowerOnState4, nothing helps. Tried cmnd/sonoff/PowerOnState 0, cmnd/sonoff/Reset 1, cmnd/sonoff / Reset 2. Re-flash and restore the configuration also heard different reviews. Please help. Need any way to disable this option

ascillato, 2018-10-08:
Hi, please, can you provide the output of the Tasmota console when you are trying the command poweronstate 0   ?

Also, please provide the output of status 0 command

ascillato2, 2018-10-08:
Hi,

You have make the exact same question in the Tasmota Chat 1 hour before posting here and has been answered there 1 minute after you asked.

There is no need to post the same question everywhere.

closed 2018-10-08 by ascillato2