arendst/Tasmota #4009: [Question] Sonoff iFan02 - deactivate sound when fan speed changes

aimartin, 2018-10-08:
I couldn't find how to do this from Tasmota, if I use the remote, there is a button that allows to mute the fan so when speed changes, the only noise you hear is the "click" from the relay, instead of a really loud BEEP (one per level, and a long beep if you turn fan off).

The problem is that, if you configure it using the remote, if AC go off (power outage or whatever), you need to push the button again, so there is no way to get rid off the remote :(

Anyone knows how can this be achieved?

Thanks!

Jason2866, 2018-10-08:
Qucik hardware solution:
Desolder the loudspeaker or buzzer

aimartin, 2018-10-08:
LOL, I was wondering if there is a less destructive way to achieve this... xD

I looked at the pictures of the circuit board and I didn't see the buzzer, although I'm not an expert...

arendst, 2018-10-08:
You must have very bad eyes...

[image]

closed 2018-10-08 by ascillato2

aimartin, 2018-10-08:
I've checked the pictures, but now in a second look, I found it:
https://user-images.githubusercontent.com/26437169/41858543-230bc352-7868-11e8-8a81-9d2f10bdc7c2.jpg

I will try to remove it... without breaking it... 

Thanks

brettlg, 2019-01-03:
I found easy solution to at least make buzzer volume more acceptable was to but a blob of hot glue on top of the buzzer speaker hole.  Buzzer still works but is not obnoxiously loud

aimartin, 2019-01-04:
I just removed it with a bit of force and a pliers xD. Seems to still be working fine...

josephhajduk, 2019-05-17:
There is a button on the remote that comes with the unit that turns off the beep.

aimartin, 2019-05-17:
Yes, but that setting is reset every time you disconnect and re-connect the iFan (for example, in a power outage), so is very annoying. 

I ended up removing the speaker and it works as expected without the beep :)

mkormendy, 2019-07-13:
Still though. We need a solution in the software settings to have this permanent upon power drop/reconnect, without removing the buzzer itself.

ShayBox, 2020-12-20:
I agree, I don't want to have to take my entire fan apart and back together to remove the buzzer, I guess I'll just tape my light switch until a software solution comes out

mkormendy, 2020-12-20:
I ended up opening up the module and removing the buzzer as well. It's completely unnecessary.

ShayBox, 2020-12-20:
I actually just figured out how to do this in software, not exactly a perfect solution but it works if you don't want to use the buzzer for alerts in things like homeassistant, you clone the ifan03 template and remap gpio10 (buzzer) to none, then apply that template to the module.

czech12, 2021-03-22:
> I actually just figured out how to do this in software, not exactly a perfect solution but it works if you don't want to use the buzzer for alerts in things like homeassistant, you clone the ifan03 template and remap gpio10 (buzzer) to none, then apply that template to the module.

Hi @Shays-Bots, I'm a Tasmota newb.  Can you explain in a little more detail what you mean by this?  I already put up my iFan02 and the buzzer is super annoying, but not as annoying as putting it in the fan was.

I went into Configure Template and selected iFan03, then selected none for GPIO10.  Everything still seemed to work after this, but the buzzer was still sounding.

Thanks!

ShayBox, 2021-03-22:
The 02 might be different, find the one labeled buzzer, and remember to Apple the template not just create it

czech12, 2021-03-22:
> The 02 might be different, find the one labeled buzzer, and remember to apply the template not just create it

GPIO10 does say buzzer on the iFan03 template, but yes I have an iFan02 and the iFan02 template does not say buzzer for any GPIO.

I think the last part is what I'm not understanding.  If I go into configure template and hit save, is that the same thing that you are referring to?  The way I'm doing it, I'm not really "creating" a template, I'm just selecting it from the list, changing GPIO10 to none, then clicking save.