arendst/Tasmota #4010: EnergyReset (?) manually via mqtt

prankousky, 2018-10-08:
I am running `Sonoff-Tasmota 6.2.1.3` on a `BlitzWolf SHP2`.

When I log in to the webinterface, I can see Energy.Total/Yesterday/Today/... I can also subscribe to these values via `device/tele/sensor` and will receive a JSON with all those values.

However, I would like to send `EnergyReset`, which should be the correct command according to wiki, manually once a day and receive those values via mqtt. Bascially, I'd like to get the daily power consumption **every day** at 23:59:59 and write it into a database.

If I just catch the output from `../tele/sensor`, I get these values once every 30 seconds (`TelePeriod 30`); I am already writing those values every 30s in a database. However, the **daily** consumption only needs to be written **once a day**.

I could grab the value via http / cmd, but I am trying not to rely on http with tasmota any longer **at all**, I think I even remember arendst mentioning somewhere (on here in the issues) that http should be used as little as possible all together.

So the easiest way for me to do this would be sending `EnergyReset` manually once a day to a specific (but **which??**) mqtt topic and then listen to another (again, which??) topic to get this value.

Or is there another way to do this? I do not want to change TelePeriod as I want 30s updates when the device is running to analyze them. But if I get an update on `../tele/sensor` every 30s, I cannot write the daily consumption to a database - at least I would not know how to do this in node-red other than via some methods I am not sure are the best practise to do so (for example, subscribing to the topic and limiting forwarded messages to once every 23h59m; if I did this, however, if the device was offline at the exact time a message would go through, I would have **no message at all** and therefor get the next once -possible already from the next day rather than the one I actually wanted to catch- whenever the device would be back online and this message would come through)

Thank you in advance for your help 👍

ascillato, 2018-10-08:
Hi,

I think that energyreset command will not do what you want.

The teleperiod was designed for that and also covers the case if the device was offline.

What you want (save once at 23:59 the value of daily consumption) will be better to be addressed in your home automation software.

closed 2018-10-08 by ascillato2

ascillato, 2018-10-08:
Questions are addressed faster in the Tasmota Discord Chat at https://discord.gg/Ks2Kzd4

prankousky, 2018-10-08:
Hi,

thank you.

Perhaps I did not explain this very well. When I log onto webgui and type `energyreset` in my console, I get this output `sonoff-7976/stat/RESULT = {"EnergyReset":{"Total":3.253,"Yesterday":0.000,"Today":0.000}}`. This is an actual output from just now. I did not consume any power through this device the last few days, but the `total` seems correct.

I was hoping there was a way to send `energyreset` via mqtt to a certain topic and then get the output from above via a different mqtt topic.

I am trying to realize this in node-red (I use this for home automation)... was just wondering if I could save some time messing with node-red flows if this could be solved merely via mqtt.

ascillato, 2018-10-08:
from wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

to send commands by mqtt:

cmnd/sonofftopic/COMMAND

so in your case should be: `cmnd/sonoff-7976/energyreset`

and the response will be as you have posted at: `sonoff-7976/stat/RESULT`, then you will need to parse the json the same as you did with telemetry.

prankousky, 2018-10-09:
@ascillato thank you very much.

Unfortunately, this did not work for me. However, I asked this question [elsewhere](https://www.reddit.com/r/nodered/comments/9mmwkg/get_mqtt_topic_payload_at_specific_time/) specific to node-red users. 

Somebody posted a solution how to get this working; this is not tasmota related, but I hope it is okay if I explain here real quick in case somebody else got here trying to find a solution:

flow 1
mqtt out -> json parser function -> json value extract funtion - set flow (or global) variable from payload
flow 2
manually inject node (i.e. at 23:59 at night) -> set payload from flow (or global) variable -> send to (for example) influxdb node