arendst/Tasmota #4012: Tasmota Compiler Error - Wemos D1 Mini

SuneBielefeldt, 2018-10-08:
**### Tasmota Compiler Error - Wemos D1 Mini**
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Development/Compiler/Upload tools used :Arduino IDE 1.8.7, new installed
- [X] Hardware used : Wemos D1 Mini
- [X] Provide the output of command ``status 0`` : Upload
```
STATUS 0 OUTPUT HERE
Sketch uses 508744 bytes (49%) of program storage space. Maximum is 1023984 bytes.
Global variables use 51772 bytes (63%) of dynamic memory, leaving 30148 bytes for local variables. Maximum is 81920 bytes.
**error: Invalid argument or value after {upload.erase_cmd} (argument #8)**

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-08:
Hi, sorry, the code in the repository is compiling fine. You can check that by the travis tag that said building pass.

Also I can compile without any issue with Arduino and also with platform.io. 

So, your issue is not Tasmota related. Sorry.

Try to delete all and reinstall again. If you search in issues you will find that this rudimentary solution has solved compiler issues.

Jason2866, 2018-10-08:
Look in the wiki: Try portable install. So you have all prepared for Tasmota.

andrethomas2, 2018-10-08:
Please follow instructions in the Wiki on how to configure your development environment correctly.

Closing this issue as it has been answered.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-08 by andrethomas2

SuneBielefeldt, 2018-10-08:
Wisely followed wiki, and performed a clean portable installation. 
Browse all WiKi, forums and google, but without finding a solution. 
That's why I've wanted help here, but there's not much understanding for that.
Hope someone can give me a useful hint.

andrethomas, 2018-10-08:
Well, the instructions are pretty basic.

Make a portable installation of arduino by downloading the zip and extracting it to a known location, lets say so that arduino.exe is inside c:\arduino\arduino.exe

Then inside c:\arduino\ create a folder named "portable"

Then fire up Arduino, go to File|Preferences and add the json url from https://github.com/esp8266/Arduino

[image]

Go to Tools|Board|Board Manager, search for ESP8266 and install it

[image]

After this clone the Tasmota repository from https://github.com/arendst/Sonoff-Tasmota to lets say c:\Sonoff-Tasmota\

Now copy contents of c:\Sonoff-Tasmota\lib into c:\arduino\portable\sketchbook\libraries\ and then copy the folder c:\Sonoff-Tasmota\sonoff into c:\arduino\portable\sketchbook\ so that you have a folder named c:\arduino\portable\sketchbook\sonoff\

Run the arduino.exe again and open sonoff.ino from the sketchbook\sonoff\ folder.

Edit what you want to edit and hit compile.

Would you like the spoon you've just been spoonfed with also?