arendst/Tasmota #4013: Adding more than 16 buttons

vaidyasr, 2018-10-08:
In RF bridge can I configure to add more than 16 buttons? The current 16 buttons is controlling only 8 switches.

andrethomas, 2018-10-08:
Sonoff RF Bridge only allows you to control 16 devices in the way you mention. To add more you need to add a second bridge.

Alternative method is to not use the buttons provided by sonoff bridge and just send rf commands via mqtt - this way you do not have the 16 device limitation.

andrethomas, 2018-10-08:
You first need to see the console output when the the bridge receives the code from your RF device, for example

`{"RfReceived":{"Sync":4880,"Low":200,"High":550,"Data":"123456","RfKey":"None"}}`

Once you have this information, you can reproduce the above with command:

`backlog RfSync 4880; RfLow 200; RfHigh 550; RfCode #123456;`

ascillato2, 2018-10-08:
Closing as the question has been answered. Thanks everyone for the contributions.

closed 2018-10-08 by ascillato2