arendst/Tasmota #4016: RF Bridge MQTT command

vaidyasr, 2018-10-09:
I am trying to switch on/off the RF buttons via MQTT without storing the button in RF bridge. I am trying the below command 
`mosquitto_pub` -t "cmnd/sonoff-bridge/Backlog" -m "RfSync 8700; RfLow 270; RfHigh 840; RfCode 4C7298" 
but the console shows below and the button is not switching on.
```
02:53:00 MQT: stat/sonoff-bridge/RESULT = {"RfSync":8700}
02:53:01 MQT: stat/sonoff-bridge/RESULT = {"RfLow":270}
02:53:01 MQT: stat/sonoff-bridge/RESULT = {"RfHigh":840}
02:53:01 MQT: stat/sonoff-bridge/RESULT = {"RfCode":4}
```

What's wrong on the above MQTT command?

When I press the button on remote I got
`02:52:05 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":8680,"Low":280,"High":840,"Data":"4C7298","RfKey": "None"}}`

Frogmore42, 2018-10-09:
Does it work if you send the commands via the web or serial console one at a time?

How do you know it is not working?

Have you tried altering the timing?

Do you have another RF receiver and logic analyzer?

I don't have an RF for Sonoff, but have done some control directly in the past.  The frequency control and timing is somewhat forgiving, but it does have to be close enough.

I found using an independent receiver and a logic analyzer made it really easy to see the difference between a working signal/pattern and one that was just a little too far from correct.

To get an extra receiver, you can either buy one for not much money or you can often borrow one from one of your RF relays.

vaidyasr, 2018-10-09:
From console also same output. The RF relay is triggering only if I send 'RfKey1' command. Is there any other method other than timing?

andrethomas, 2018-10-09:
The method seems correct. I did this with a 433mhz doorbell button so if I send the code back in the same way the doorbell rings.

Some receivers will be more sensitive to the RFsync, RFlow,RFhigh values.

I recommend doing a couple of receives on the RF bridge to see how much they differ and find a happy middle (would usually be a round number) - Likely in your case 8700,300,850.

Some receivers also expect more than one so you could try:
---
backlog RfSync 8700; RfLow 270; RfHigh 840; RfCode #4C7298;RfSync 8700; RfLow 270; RfHigh 840; RfCode #4C7298;
---

also

Jason2866, 2018-10-09:
Hello @vaidyasr 
If your device doesnt work with standard commands RfSync RfLow RfHigh RfCode you have to use RfRaw. I have had a device too, that doesnt work with this commands. With RfRaw i got it working.
For Details using Command RFRaw this here is the wrong place since it is not Tasmota related.
Feel free to ask at Portisch GitHub  https://github.com/Portisch/RF-Bridge-EFM8BB1
he is the developer of the custom firmware of the RF Chip and the command RfRaw.

andrethomas2, 2018-10-09:
Closing this issue as it has been answered.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-09 by andrethomas2

Geoff571, 2018-10-12:
I have exactly the same issue. 

I've used rfkey<x> 2 to record the RF codes and tested that working fine. Using RfKey<x> 5 to display the associated data, then used BACKLOG to play in the RfSync, RfLow, RfHigh and RfCode data.
This does not have the same effect as triggering the switch using the RfKey<x> command.

**Works**:
`22:04:04 RSL: Group 0, Index 2, Command RFKEY, Data 
22:04:04 MQT: stat/sonoffrfbr/RESULT = {"RfKey2":"Learned sent"}
22:04:05 MQT: tele/sonoffrfbr/RESULT = {"RfRaw":{"Data":"AAA055"}}
22:04:25 CMD: rfkey2 5
22:04:25 SRC: WebConsole from 192.168.1.108
22:04:25 RSL: Group 0, Index 2, Command RFKEY, Data 5
22:04:25 MQT: stat/sonoffrfbr/RESULT = {"RfKey2":{"Sync":10750,"Low":350,"High":1030,"Data":"144015"}}`

**Doesn't work:**

`22:05:49 RSL: Group 0, Index 1, Command BACKLOG, Data RfSync 10750; RfLow 350; RfHigh 1050; RfCode 144015
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFSYNC, Data 10750
22:05:50 MQT: stat/sonoffrfbr/RESULT = {"RfSync":10750}
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFLOW, Data 350
22:05:50 MQT: stat/sonoffrfbr/RESULT = {"RfLow":350}
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFHIGH, Data 1050
22:05:50 MQT: stat/sonoffrfbr/RESULT = {"RfHigh":1050}
22:05:50 SRC: Backlog
22:05:50 RSL: Group 0, Index 1, Command RFCODE, Data 144015
22:05:51 MQT: stat/sonoffrfbr/RESULT = {"RfCode":144015}
22:05:51 MQT: tele/sonoffrfbr/RESULT = {"RfRaw":{"Data":"AAA055"}}`

Has anyone got this working in this manner? Am I missing something?

Geoff571, 2018-10-21:
@vaidyasr I'm having this issue too, but not tried to resolve it yet as I need to solve my other issue above first.

andrethomas, 2018-10-21:
Works fine for me and I cannot see anything obvious wrong with yours - perhaps you did not get enough samples from origin to get valid values - I know this method works because I use it to make a doorbell ring.