arendst/Tasmota #4017: HC-SR04  & Sleepmode

nosaj66au, 2018-10-09:
Hi Guys,
   I'm looking for some information on the relationship between Sleepmode (250ms) with a HC-SR04 and if the Telemetry time is "300".
 I'm now using a build that support sleepmode however I would like to know when will the HC-SR04 be polled to take a measurement? All the time, just when not sleeping, every 300 (Telemetry time) sec or ????

Thanks in advance. 
Jason

benzino77, 2018-10-09:
The code of sensor [xsns_22_sr04.ino](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_22_sr04.ino) suggest that the measurement is performed when:
1. callback FUNC_JSON_APPEND is invoked
2. callback FUNC_WEB_APPEND is invoked

The information on the wiki [Sensor-API](https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-API#func_json_append) inform that first callback is called when teleperiod  is due to append telemetry data, and the second callback is called every 2.5 sec (when HTML code should be added to the Tasmota web-interface). 
There is information in the [changelog](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/_changelog.ino) that the new version brings something called `WebRefresh` which I think can change those 2.5 sec mentioned above.

ascillato2, 2018-10-09:
Closing as the question has been answered. Thanks everyone for their contributions.

closed 2018-10-09 by ascillato2