arendst/Tasmota #4019: Feature request: display Running Hours for Pow

i3laze, 2018-10-09:
**Have you look for this feature in other issues and in the wiki?**
Yeap
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
Energy Total doesn't give a clue about total running time.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
I've got a heater with variable power consumption, triggered by some scripts or manually.
I'd love to see Runtime Today, Runtime YesterDay and Runtime Total alongside with Energy outputs on Sonoff's main web-page.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Currently, I'm tinkering with aggregating such counters on Domoticz.. but out of the box solution might be better.

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-09:
Hi, duplicated. See #3971

closed 2018-10-09 by ascillato2

i3laze, 2018-10-09:
Sorry, thanks.  The other issue description was kinda messy. :(

ascillato, 2018-10-09:
hehehe, no problem. Please, just add there your ideas.

i3laze, 2018-10-30:
#3971 solved only it’s own demand for a total start time..

Personally, I'd love to see:

Runtime Today,
Runtime Yesterday
Runtime Total

Unlike a fridge, i've got a heater triggered by Sonoff directly.
For a 24x7 fridge a Runtime value might be calculated for power consumption bigger than some threshold.