arendst/Tasmota #4020: Feature Request: Trim Spaces after Rfraw command before command execution

Jason2866, 2018-10-09:
**Have you look for this feature in other issues and in the wiki?**
yes
**Is your feature request related to a problem? Please describe.**
If data string after RfRaw includes spaces command execution fails.
`RfRaw AA B0 3C 04 14 01E1 042F 0BE0 1C70 2301101001011010011001010101101010011010011010010123011010010101011001100101101010010101011010100101 55 `
doesnt work
**Describe the solution you'd like**
delete spaces automatically when command is executed to avoid unnecessary errors

**Describe alternatives you've considered**
none

andrethomas, 2018-10-13:
Proposed PR https://github.com/arendst/Sonoff-Tasmota/pull/4053

andrethomas2, 2018-10-14:
Closing this issue as it the PR made for this request has been merged into development.

closed 2018-10-14 by andrethomas2