arendst/Tasmota #4021: Power consumption and LED blinks if wifi is down - Implement Wifi down handling

madexx, 2018-10-09:
Normally i switch off wifi at night. In this case the LEDs blinks the whole night. I also noticed that the power consumption is increasing up to the same level with sleep = 0.

It would be great, if it would be possible to switch off the LEDs after a certain time and try to reconnect every 5minutes for example.

ascillato, 2018-10-09:
Type in the console

SetOption31 0

Please, read the wiki for more infomation. Thanks

closed 2018-10-09 by ascillato2

madexx, 2018-10-09:
Ok, thank you. But i still have the issue with the power consumption. It increases and is "high", also when the wifi is reconnected. And the only way to get it down again is to unplug an plug it again. A softreset does not help.