arendst/Tasmota #4022: Compiler error on core 3.0.0

andrethomas, 2018-10-09:
@arendst 

**Describe the bug**
```
C:\arduino-1.8.7\portable\sketchbook\sonoff\sonoff.ino: In function 'void Every250mSeconds()':

C:\arduino-1.8.7\portable\sketchbook\sonoff\sonoff.ino:2040:77: warning: 't_httpUpdate_return ESP8266HTTPUpdate::update(const String&, const String&)' is deprecated (declared at C:\arduino-1.8.7\portable\packages\esp8266\hardware\esp8266\2.4.3\libraries\ESP8266httpUpdate\src/ESP8266httpUpdate.h:87) [-Wdeprecated-declarations]

           ota_result = (HTTP_UPDATE_FAILED != ESPhttpUpdate.update(mqtt_data));
```

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : 
- [x] _Development/Compiler/Upload tools used_ : Arduino 1.7 w/ 2.4.3(hehe) core
- [x] _Provide the output of command_``status 0`` :
```
18:26:13 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:26:13 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:28:52","StartupUTC":"2018-10-09T16:57:21","Sleep":0,"BootCount":9,"SaveCount":21,"SaveAddress":"3FA000"}}
18:26:13 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.13","BuildDateTime":"2018-10-09T18:51:53","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
18:26:13 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wireless",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
18:26:13 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":520,"Free":3572,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000809","8FDAE394","000003A0","23B617CE","000003C0"]}}
18:26:13 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4052","IPAddress":"192.168.42.42","Gateway":"192.168.32.1","Subnetmask":"255.255.240.0","DNSServer":"192.168.32.1","Mac":"5C:CF:7F:10:2F:D4","Webserver":2,"WifiConfig":5}}
18:26:13 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_102FD4","MqttUser":"DVES_USER","MqttType":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:26:13 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 09 17:26:13 2018","Local":"Tue Oct 09 18:26:13 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:01","Sunset":"18:13"}}
18:26:13 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-09T18:26:13"}}
18:26:13 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-09T18:26:13","Uptime":"0T00:28:52","Vcc":2.989,"Wifi":{"AP":1,"SSId":"Wireless","BSSId":"DE:AD:0T:AS:MO:TA","Channel":3,"RSSI":96}}}
```

**To Reproduce**
Install staged core, and hit compile

**Expected behavior**
Neither me or I like warnings ;)

arendst, 2018-10-09:
I have not touched stage in ages but considering the sleep solution I might....

andrethomas, 2018-10-09:
;)

It does not look like they will release 2.5.0 until December so that's ok I guess.

andrethomas, 2018-10-09:
I just did a jippo on @ascillato fork to make a json for 2.4.3 (which technically will never exist, but anyway)

arendst, 2018-10-09:
???
Urban Dictionary: A 'Jippo'is someone who lives in a caravan on other peoples land without permission.
???

Jason2866, 2018-10-09:
I have no compiler warnings :-) with stage 3.0 and it works perfect. Sleep :-)
and very responsive.
[image]

andrethomas, 2018-10-09:
> ???
> Urban Dictionary: A 'Jippo'is someone who lives in a caravan on other peoples land without permission.
> ???

LOL

Around here it means you haxored something together to make it work... like using duct tape to keep your rear view mirror in place... or if you like, using duct tape for underpants ;)

ascillato2, 2018-10-09:
I will look into it.

andrethomas, 2018-10-10:
@ascillato Sure, should I make the json update from your repo or direct from [https://github.com/esp8266/Arduino](https://github.com/esp8266/Arduino)

andrethomas, 2018-10-10:
Interesting 

[image]

Experimental json for Arduino board manager for 2.5.0 may be used as
`http://thehackbox.org/package_esp8266com_index.json`

andrethomas, 2018-10-10:
Also, it seems that warning does not appear in Arduino 1.8.5

ascillato, 2018-10-11:
The warning of deprecated function cames from:

https://github.com/esp8266/Arduino/blob/d17ffc287493e76e7d6895c5c623ff4ae66baaa6/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L83-L108

The new example of the library uses `ESPhttpUpdate.update` as:

https://github.com/esp8266/Arduino/blob/d17ffc287493e76e7d6895c5c623ff4ae66baaa6/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino#L55

So, the syntax in sonoff.ino

https://github.com/arendst/Sonoff-Tasmota/blob/ac595d44b04730ca3d8a2ac1bd70741c0096ecf5/sonoff/sonoff.ino#L2040

Should be:

`ota_result = (HTTP_UPDATE_FAILED != ESPhttpUpdate.update(EspClient, mqtt_data));`

Tested. Ok.

Made PR #4032 for using the new syntax from core STAGE and core 2.5.0 and up.

closed 2018-10-11 by ascillato2

ascillato, 2018-10-11:
The STAGE Core has a bug when doing an OTA from an URL. If uploading a file from the webmenu, it works fine.

Issue reported at https://github.com/esp8266/Arduino/issues/5023