arendst/Tasmota #4023: Shelly 1 random switching

dunnink, 2018-10-09:
Hello all!

I myself have two Shelly 1's. One I have flashed with the Tasmota firmware by using PlatformIO and this tutorial: https://youtu.be/sIVjeYZeLNU; Since I'm having trouble with the first one I wait with flashing the second one.

I made the by 'TheSmartHomeHookUp' suggested changes to the sonoff.ino file regarding the pullup resistors. Having flashed the firmware, the sonoff perfectly pops up as an access point to which I connect succesfully. I can select the Shelly from the profile list. And then I ran into the problem.

It seems the switch input is hovering and causing random switch behaviour. Sometimes the relays switches twice within 5 seconds. It switches all the time, no matter if I deselect it as a Shelly device. MQTT is not configured yet so it can't be due to that. Attached is some console logging.
```
22:00:30 RSL: stat/sonoff/RESULT = {"POWER":"ON"} 
22:00:30 RSL: stat/sonoff/POWER = ON
22:00:40 MQT: Attempting connection...
22:00:41 DNS: Query done. MQTT services found 0
22:00:41 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
22:00:48 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
22:00:48 RSL: stat/sonoff/POWER = OFF
22:00:52 MQT: Attempting connection...
22:00:53 DNS: Query done. MQTT services found 0
22:00:53 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
22:00:59 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
22:00:59 RSL: stat/sonoff/POWER = ON
22:01:02 APP: Serial logging disabled
22:01:02 RSL: tele/sonoff/STATE = {"Time":"2018-10-09T22:01:02","Uptime":"0T00:10:42","Vcc":3.214,"POWER":"ON","Wifi":{"AP":1,"SSId":"-------v10","RSSI":62,"APMac":"xx:xx:xx:xx:xx:xx"}}
22:01:02 RSL: tele/sonoff/SENSOR = {"Time":"2018-10-09T22:01:02","Switch1":"ON"}
22:01:02 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
22:01:02 RSL: stat/sonoff/POWER = OFF
22:01:03 MQT: Attempting connection...
```

I am shure that it has to do with the Tasmota firmware because I don't see the behaviour with the stock Shelly firmware. As I already mention I can see the behaviour even without having the Shelly selected as the switch, but left at default (Sonoff Basic). I do not see the behaviour when I power the Shelly from the FTDI converter, hinting me on some configuration for the GPIO that is attached to the switch input.

Do you have any suggestions on how to fix this?

```
22:08:56 RSL: stat/sonoff/STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:08:56 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:18:36","StartupUTC":"2018-10-09T20:50:20","Sleep":0,"BootCount":8,"SaveCount":70,"SaveAddress":"F6000"}}
22:08:56 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-10-09T22:40:57","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:08:56 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
22:08:56 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":491,"Free":512,"Heap":13,"ProgramFlashSize":1024,"FlashSize":2048,"FlashMode":3,"Features":["00000809","1FDAE594","24000010","23B617CE","00000000"]}}
22:08:56 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1912","IPAddress":"192.168.2.125","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"","Webserver":2,"WifiConfig":5}}
22:08:56 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5BA778","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:08:56 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 09 21:08:56 2018","Local":"Tue Oct 09 22:08:56 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:01","Sunset":"18:13"}}
22:08:56 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-09T22:08:56","Switch1":"ON"}}
22:08:56 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-09T22:08:56","Uptime":"0T00:18:36","Vcc":3.247,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":56,"APMac":""}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

closed 2018-10-09 by ascillato2

JDunnio, 2018-10-10:
Tnx, that looks promising. I'm going to start all over.

If Theo or Adrian lives close by the Netherlands I would be happy to supply a Shelly 1 if it helps the development.

dunnink, 2018-10-10:
Thumbs up! By using the development version it was fixed.