arendst/Tasmota #4024: Hue emulation not fully working on 6.2.1.9

SoundGuy, 2018-10-10:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
I have tasmota running on a nodeMCU with a switch connected to D7 and a LED lamp connected to D8 and configured with PWM.
With version 6.2.1.  My alexa communicated with Hue bridge emulation. with 6.2.1.9 it only displays the brightness and state but can't send commands and change state or brightness. When i reverted back to 6.2.1. release it started working again.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ :  NodeMCU v.1
- [X ] _Development/Compiler/Upload tools used_ : platform io
- [ ] _Provide the output of command_``status 0`` :
```1
this is from the 6.2.1  version, I will need to reupload 6.2.1.9 to provide that. 

2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS11Qos: 1{"StatusSTS":{"Time":"2018-10-10T00:59:31","Uptime":"0T00:09:11","Vcc":2.818,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"corbomite5-2.4","RSSI":100,"APMac":"9C:6F:52:29:54:69"}}}
2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS10Qos: 1{"StatusSNS":{"Time":"2018-10-10T00:59:31","Switch1":"ON"}}
2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS7Qos: 1{"StatusTIM":{"UTC":"Tue Oct 09 23:59:31 2018","Local":"Wed Oct 10 00:59:31 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:40","Sunset":"16:14"}}
2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS6Qos: 1{"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_864075","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS5Qos: 1{"StatusNET":{"Hostname":"sonoffJacuzzi-0117","IPAddress":"192.168.1.28","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:86:40:75","Webserver":2,"WifiConfig":3}}
2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS4Qos: 1{"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS3Qos: 1{"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["corbomite5-2.4",""],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
2018-10-10 02:59:30Topic: stat/sonoffJacuzzi/STATUS2Qos: 1{"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
2018-10-10 02:59:29Topic: stat/sonoffJacuzzi/STATUS1Qos: 1{"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:10","StartupUTC":"2018-10-09T23:50:20","Sleep":0,"BootCount":38,"SaveCount":370,"SaveAddress":"F8000"}}
2018-10-10 02:59:29Topic: stat/sonoffJacuzzi/STATUSQos: 1{"Status":{"Module":18,"FriendlyName":["SonoffOded-Jacuzzi-NodeMCU"],"Topic":"sonoffJacuzzi","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

```

**To Reproduce**
_Steps to reproduce the behavior:_
installed 6.2.1.9 on a nodemcu
enable hue emulation
connect with Alexa
try to turn lights on and off . doesn't work

**Expected behavior**

expeted to lights to be turned on and off like they do in 6.2.1.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

SoundGuy, 2018-10-10:
how do i find out in Platform IO?

SoundGuy, 2018-10-10:
this is the relevant line from the .ini file:

[common] ; ************************************************************
; *** Esp8266 core for Arduino version 2.3.0
platform = espressif8266@1.5.0

it's the only one unmarsked but then i think it gets override with this one:

[env:wemos-d1-mini]
platform = espressif8266

ascillato, 2018-10-10:
In platformio.ini file.

Please, should be better if you use the precompiled bins from development at http://thehackbox.org/tasmota/

ascillato, 2018-10-10:
If you update by OTA with those precompiled bins, your config will remain in your device and we can test if your issue is wrong core or not.

SoundGuy, 2018-10-10:
ok. lemme try

ascillato, 2018-10-10:
Otherwise, use the precompiled bins from release

closed 2018-10-10 by ascillato2