arendst/Tasmota #4025: Use a different font for O-LED display

mike2nl, 2018-10-10:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
Nope, because all are sleeping right now ;-) and the advise 2 weeks ago was to ask gemu. Yeah gemu did can't be reached right now. Possible he is very, very busy with new things. So i'm here now.
- [ ] Development/Compiler/Upload tools used :
Not needed because it's a question
- [x] Hardware used :
0.96 inch yellow/blue O-LED via I2C on a sonoff sc 
- [ ] Provide the output of command ``status 0`` :
Not needed here because it's a question

- It's about the comabination of degress Celsius °C and it's possible a dummy question but here we are.

I was thinking about the degrees ascii character '0xB0' for temperature values and i could use some other characters too e.g. W/m2 (uppercase 2). That 0xB degree char is defined in one of the font files under the Adafruit-GFX-Library-1.2.9 library.
- Font file name: TomThumb.h, the only who have it
- #define TOMTHUMB_USE_EXTENDED 1 is set
I would use that one but i understand not 100% what the hint means in the readme for the Adafruit-GFX-Library-1.2.9.

Hint:
'Fonts' folder contains bitmap fonts for use with recent (1.1 and later) Adafruit_GFX. **= clear**
To use a font in your Arduino sketch, \#include the corresponding .h file **= clear**
and pass address of GFXfont struct to setFont(). **= not clear**
Pass NULL to revert to 'classic' fixed-space bitmap font. **= not clear**

My thoughts are that the include must be placed in the drv_13_display.ino driver file. But i did not 
find the right place to add the font to the GFXfont struct.

And how to stop the Pass NULL, i did not find it: Pass NULL to revert to 'classic' fixed-space bitmap font

It's really early today and possible that's the issue why i can't see it right now. So every help is very
appreciated right now because i stay again in the front of all these left trees in my garden.

arendst, 2018-10-10:
The current way Tasmota is handling these non-standard codes is depending on display hardware specific font supported code.

Currently only the degrees code is supported keeping in mind the limited fonts of a cheap lcd display.

See xdrv_13_display.ino where the ~ is replaced by a display driver supported degree code:
```
char* DisplayLogBuffer(char temp_code)
{
  char* result = NULL;
  if (disp_log_buffer_cols) {
    if (disp_log_buffer_idx != disp_log_buffer_ptr) {
      result = disp_log_buffer[disp_log_buffer_ptr];
      disp_log_buffer_ptr++;
      if (DISPLAY_LOG_ROWS == disp_log_buffer_ptr) { disp_log_buffer_ptr = 0; }

      char *pch = strchr(result, '~');  // = 0x7E (~) Replace degrees character (276 octal)
      if (pch != NULL) { result[pch - result] = temp_code; }
    }
  }
  return result;
}
```
Search the code for examples of the above function call.

mike2nl, 2018-10-10:
@arendst
Thanks for the hint, and yes i found it and will test it.
That will help me out to find or solve things.

I will close the question now and will possible make a PR with two or three additions.

Thanks again Theo.