arendst/Tasmota #4026: Sonoff TH-16 and ds18b20, very high display delay.

lev2k, 2018-10-10:
Sonoff TH-16 - Tasmota 6.2
hello, I have noticed that the "ds18b20 " sensors seem have a very high "display delay".
Test: A water bowl with about 5 litres of water.
The water was heated by an immersion heater until 70° was displayed in the Tasmota Web-UI.
The immersion heater was removed IMMEDIATELY, but the displayed temperature continued to rise by about 3° within a good minute (or probably a bit more).
This would mean a display delay of about one minute?
Is there any way to reduce this?

thanks in advance

arendst, 2018-10-10:
Could be many reasons.
- What's the hysteresis of your sensor - how quick can it react on temperature changes
- Latest version of Tasmota probe the sensor every 2 seconds
- The web interfaces default updates it's screen every 2.345 second
- Depending on teleperiod settings you will see JSON data

So I don't think Tasmota is the source of your issue but rather the slow response time of your sensor.

lev2k, 2018-10-10:
thanks for your reply.
I'm using a sensor like this one:
https://www.amazon.co.uk/Sonoff-Waterproof-Temperature-Wireless-Compatible/dp/B06XW6STL7
Is it really possible, that the response time of this sensor is **so** bad, that it results in that huge "one-minute" delay?
Any recommendations for a better one (waterproof), with faster response time?

ascillato2, 2018-10-10:
Hi, closing as the questions has been answered and that it is not a Tasmota software issue. Thanks

closed 2018-10-10 by ascillato2