arendst/Tasmota #4027: Adding support for RGB/WW/CW Controller with music and timer mode

Damontir, 2018-10-10:
Hey!

Just bought a 
"DC12V 24V Wifi LED Controller RGB/RGBW/RGBWW 16 Million Colors Music and Timer Mode Wifi " 
because i have some meters RGBWW leds laying around here .

I found out that its driven by a ESP8285

Maybe worth a look to get this little baby running?

https://www.aliexpress.com/item/DC12V-24V-Wifi-LED-Controller-RGB-RGBW-RGBWW-16-Million-Colors-Music-and-Timer-Mode-Wifi/32674135296.html

[image]
[image]

ascillato, 2018-10-10:
Hi

You will need to find which gpios are being used and then set the module type as generic.

ascillato2, 2018-10-10:
If you can find the GPIOs that are being used, please, share it and we can add that information to the wiki.

Thanks.

closed 2018-10-10 by ascillato2