arendst/Tasmota #4039: LoRa / LoRaWan support

gabrielklein, 2018-10-12:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
A way to be able to communicate with a sonoff (or custom board) using the LoRa communication channel.

Example:
https://www.aliexpress.com/item/ESP32-OLED-Display-Bluetooth-WIFI-Lora-32-Module-IOT-Development-Board-module/32914396758.html

**Describe alternatives you've considered**
Doing everything myself. :)

**Additional context**
None

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-12:
Hi,

The actual scope of Tasmota is Home Automation by wifi and mqtt. How LoRa fits there? What are your ideas and use case for a sonoff device with LoRa extra hardware?

ascillato, 2018-10-12:
I add this just for reference.

Those are SPI devices. The data sent/received is similar to the RF Bridge
LoRa (LongRange) is a IoT communication channel intended for very low consumption sensors in a wide area (like a field or a city)

Library: https://github.com/weustace/LoRaLib
Datasheet: http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf
Devices:
* https://www.aliexpress.com/item/433Mhz-Lora-SX1278-Long-Range-RF-Wireless-Module-SPI-DRF1278F-For-Arduino/32795077709.html
* https://www.aliexpress.com/item/RFM98W-RFM98W-Wireless-transceiver-modules-LoRa-spread-spectrum-communications-433M-SX1278-16-16mm/32707134262.html

Frogmore42, 2018-10-12:
A LoRa to MQTT gateway makes sense and a LoRa sensor makes sense, but it seems like a sensor that has both makes much less.

There appears to be an existing project for the gateway 
https://www.loraserver.io/

If you want a sensor, this might help
https://github.com/sandeepmistry/arduino-LoRa

ascillato, 2018-10-13:
@gabrielklein

What are your ideas and use case for a sonoff device with LoRa extra hardware?

gabrielklein, 2018-10-13:
Hi, I have a few SX12xx board.

I have some birds but they are too far from my house and internet connection.

I want to have a few sensors:
- Temperature of water
- Temperature of the air
- Water heater
- Air heater
- Weight of Seeds

Some rules:
- Switch on "water heater" if temperature is <0°c
- Switch on "air heater" if temperature is <0°c
- Switch on / off some light
- Report temperature and energy usage
- Report weight of seeds (alert if too low)

ascillato2, 2018-10-13:
So, your idea is that Tasmota reads LoRa as a bridge to mqtt like the sonoff bridge. Nice.

jimmys01, 2018-11-13:
I am also very interested to tasmota supporting RFM69 and/or LoRa radios. This is huge for battery powered sensors that only report once every 10 minutes or once per day etc.

jimmys01, 2018-11-13:
The esp8266 will act like a gateway for the LoRa. The nodes dont matter, they can be esp8266 or arduino based. The arduino based are perfect for battery life. The esp8266 node can also be very usefull for deployments that have power available but no internet access near by. Imagine it like  a sonoff RF but for distances up to several kilometers

Jason2866, 2019-01-04:
So for Tasmota there would be a driver needed for a LoRa receiver to switch relais on / off?
So why not build an simple LoRa receiver device that triggers a GPIO from Sonoff to switch it?
IMHO overkill (needed resources) for Tasmota would be if a LoRa Transceiver should be added to a Tasmota device to transmit values from Sensors connected to. 
For this use cases there are LoRa Sensors available.
@jimmys01  Your comparison with sonoff RF is just like that. Just receiving signal to switch not more.

Cloolalang, 2019-05-21:
https://github.com/1technophile/OpenMQTTGateway/releases

Jason2866, 2019-05-21:
Since there is a possibility to add a LoRa 2 mqtt gateway this feature request can be closed and nobody is working on implementing on a PoC.

gemu2015, 2019-08-07:
i have 2 gateways in the works, would be nice if somebody would help me to test them

1. iBeacon gateway with hm17 ble chip scans for ibeacon key fobs and transmits rssi 
2. Moritz gateway for EQ3 MAX devices (window contacts, pushbuttons, and thermostats)

simply copy over my drivers from my fork =>https://github.com/gemu2015/Sonoff-Tasmota
(no libs needed)

xsns_92_ibeacon.ino
xsns_97_moritz.ino

ascillato, 2019-08-07:
@gemu2015 

Cool !!!

Please, can you make a pull request? Thanks.

meingraham, 2019-08-07:
@gabrielklein - are you still interested?

@gemu2015 you may also want to post a message in the Tasmota Discord server. Andreas Spiess (the guy with the Swiss accent) seems to be a heavy LoRa user... and he uses Tasmota. Maybe he'd be interested.

Mike

gemu2015, 2019-08-07:
@ascillato 
i may pr the ibeacon driver. it is extremely simple and the hardware is easy to adapt and it works stable. (only rec,trx ) it claims the serial port.
you need a hm17 ble4 module and the above ble 4 beacons.

the moritz driver is not complete yet and it is extremely complicated. the max devices are bidirectional and have to be paired and messages have to be acknowledged.
what works is automatic pairing of all devices (when in learning mode) and reporting all states.
this is complete for pushbuttons and window contacts. (they only report states)
but the thermostats are much more complex. (what works is setting the thermostat temperatures by mqtt command)
however the thermostats have a bunch of presets that must still be implemented
(e.g. time table of preconfigured temperatures for every weekday)
i am working on a web ui and mqtt for that.
also this driver needs a 24c256 i2c eeprom to store all the data needed for up to (50 devices) 
the CC1101 is connected via hardware spi.
the max solution is a proprietary solution even the protocol had to be reverse engineered.
the CC1101 driver for the max devices comes from Rudolf Koenigs culfw project.

@meingraham 
i have currently no solution for other lora sensors that are needed by @gabrielklein (temperature, etc or even actors, it would also be extremely important that at least actors should be bidirectional)

gabrielklein, 2019-08-07:
@meingraham I have too many interests and not enough time :) 

Tasmota as LoRa Gateway: I'm looking more in a LoRaWan solution these days. The price of gateways goes down.
Tasmota as LoRa client: Using LoRa instead of WiFi to send data definitely makes sense. As an example a Pow2 module could send data every x minutes over LoRa.

@gemu2015 Thank you for the beacons, I should receive them in 2-3 weeks...

ascillato2, 2019-10-11:
Closing this issue as initial support has been added by @gemu2015 on PR #3309.

Thanks. :+1: 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-11 by ascillato2

MarkTyr, 2022-01-14:
How can i use the Lora support in Tasmota? 

MQTT -> ESP32 with Tasmota -> LORA -> ESP32 with Tasmota -> MQTT

ascillato, 2022-01-14:
Hi, there was a confusion in this issue. The PR linked is not Lora. It is just an I2C sensor support.

So far, LORA is not supported in Tasmota. Any PR is very welcome.

mcarbonneaux, 2022-09-18:
esp32 with lora integrated:
https://fr.aliexpress.com/item/32846302183.html

c64emulator, 2022-10-19:
> Hi, there was a confusion in this issue. The PR linked is not Lora. It is just the support for a sensor.
> 
> So far, LORA Gateways are not supported in Tasmota. Sorry. Any PR is very welcome.

This feature request is closed? Should this be reopened?

barbudor, 2022-10-19:
Hi
For feature requests, the suggested process is now to open a Discussion
So would be cool if you start a new Discussion and describe your use case.
If we can grab use cases and come to spec the feature, someone might pickup the task

c64emulator, 2022-10-19:
Here comes my use case for a ESP32 with LoRa (e.g. HELTEC LoRa32) on tasmota:
I want to read the energy consumption in our house. In germany (where we leave) the good old ferrari counters are replaced with new digital electricity meters. This meters have a IR interface with some specific protocol (sml). Tasmota provides support for this protocol. So far so good...
BUT: This meters are placed somewhere in the house, where WiFi is not available (cellar, roof) or in metal cabinets where rf does not come through. Another use case: a friend is living in a neighbourhood, where the meters of all houses are located in a room 200m from its own house. 
Therefore I would give LoRa a chance.
My idea:
electricity meter with IR sender -> Ir diode head -> ESP32 with Tasmota (for decoding) -> LoRA (for transmission) -> ESP32 with Tasmota (as receiver) -> MQTT

barbudor, 2022-10-19:
Don't duplicate posts
The suggestion was to open a new discussion, not to spread the same text across multiple threads